MarlinFirmware/Marlin #17123: Board RAMPS 1.4 + Standart Smart Display 4x20. The display cannot display more than 1000. The mechanics of the printer are 1.5 meters. Firmware Marlin 2. Where can I change the code to display more than three digits?

Evgeniy-U, 2020-03-11:
[image]

ellensp, 2020-03-11:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/HD44780/ultralcd_HD44780.cpp#L833 and line 835  change ftostr4sign to ftostr5sign.

closed 2020-03-11 by thinkyhead

thinkyhead, 2020-03-11:
This might also help:
```diff
  const char* ftostr4sign(const float &f) {
    const int i = (f * 100 + (f < 0 ? -5: 5)) / 10;
-   if (!WITHIN(i, -99, 999)) return i16tostr4signrj((int)f);
+   if (!WITHIN(i, -999, 9999)) return i16tostr4signrj((int)f);
```

thinkyhead, 2020-03-11:
Actually, it appears from my testing that `ftostr4sign` already prints values from 1000-9999 with 4 digits. Perhaps they are being overlapped in the printing.