MarlinFirmware/Marlin #17125: GCODE_MACROS (i.e. M810 - M819) only execute once

tmartin007, 2020-03-11:
When defining a macro with M810, e.g. (a very simple example):

M810 G0 X30 | M400 | Y30 | M400 | Z30 | M400

it will only execute completely on it's first invocation (i.e. M810).
On any subsequent invocation, only the first command (i.e. G0 X30) 
will be executed.

Steps to Reproduce:

1) Use Marlin 2.0.4.4 on a BigTree SKR Pro 1.1V with  "GCODE_MACROS" enabled in Configuration_adv.h (zipped configuration files attached) 
2) define a macro like this: "M810 G0 X30 | M400 | Y30 | M400 | Z30 | M400"
3) invoke the macro by issuing "M810"
4) observe the correct execution of the macro
5) issue e.g. "G0 X1 Y1 Z1" and observe all 3 axis to move to (e.g.) "1"
6) invoke the macro again by issuing "M810"
7) observe that only the first command of the macro (i.e. "G0 X30") is executed
8) repeat 5) to 7) as often as you like

tmartin007, 2020-03-11:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4319851/Configuration.zip)

thinkyhead, 2020-03-11:
@tmartin007 — Thanks for the report! I can immediately guess why this is happening. I'll post a patch shortly.

thinkyhead, 2020-03-11:
Ok, should be working now!

closed 2020-03-11 by thinkyhead

tmartin007, 2020-03-12:
I do confirm - this is fixed and works beautifully now. 
Thank you very much!

ps: Looking forward to your comments on #17128

commit 318a1663 (2020-03-11): Fix M810 macro multiple use

Fixes #17125