MarlinFirmware/Marlin #17126: Latest bugfix doesn't compile for mks robin nano

porlock, 2020-03-11:
### Bug Description

Cant compile
[vanila.zip](https://github.com/MarlinFirmware/Marlin/files/4320694/vanila.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1) try to compile 

**Expected behavior:** [What you expect to happen]
Compilation successful 

**Actual behavior:** [What actually happens]
Crash 
In file included from Marlin\src\sd\../inc/MarlinConfig.h:44:0,
                 from Marlin\src\sd\cardreader.cpp:23:
Marlin\src\sd\cardreader.cpp: In function 'void announceOpen(uint8_t, const char*)':
Marlin\src\sd\../inc/../core/utility.h:75:41: error: passing 'const restorer<signed char>' as 'this' argument discards qualifiers [-fpermissive]
 #define RESTORE(N) restorer_##N.restore()
                                         ^
Marlin\src\sd\../inc/../core/serial.h:53:33: note: in expansion of macro 'RESTORE'
   #define _PORT_RESTORE(n)      RESTORE(n)
                                 ^~~~~~~
Marlin\src\sd\../inc/../core/serial.h:67:33: note: in expansion of macro '_PORT_RESTORE'
 #define PORT_RESTORE()          _PORT_RESTORE(1)
                                 ^~~~~~~~~~~~~
Marlin\src\sd\cardreader.cpp:459:5: note: in expansion of macro 'PORT_RESTORE'
     PORT_RESTORE();
     ^~~~~~~~~~~~
Marlin\src\sd\../inc/../core/utility.h:71:15: note:   in call to 'void restorer<T>::restore() [with T = signed char]'
   inline void restore() { ref_ = val_; }
               ^~~~~~~
Marlin\src\sd\cardreader.cpp: In static member function 'static void CardReader::openFileRead(char*, uint8_t)':
Marlin\src\sd\../inc/../core/utility.h:75:41: error: passing 'const restorer<signed char>' as 'this' argument discards qualifiers [-fpermissive]
 #define RESTORE(N) restorer_##N.restore()
                                         ^
Marlin\src\sd\../inc/../core/serial.h:53:33: note: in expansion of macro 'RESTORE'
   #define _PORT_RESTORE(n)      RESTORE(n)
                                 ^~~~~~~
Marlin\src\sd\../inc/../core/serial.h:67:33: note: in expansion of macro '_PORT_RESTORE'
 #define PORT_RESTORE()          _PORT_RESTORE(1)
                                 ^~~~~~~~~~~~~
Marlin\src\sd\cardreader.cpp:520:5: note: in expansion of macro 'PORT_RESTORE'
     PORT_RESTORE();
     ^~~~~~~~~~~~
Marlin\src\sd\../inc/../core/utility.h:71:15: note:   in call to 'void restorer<T>::restore() [with T = signed char]'
   inline void restore() { ref_ = val_; }
               ^~~~~~~
Compiling .pio\build\mks_robin_nano\src\src\sd\usb_flashdrive\lib-uhs2\masstorage.cpp.o
*** [.pio\build\mks_robin_nano\src\src\sd\cardreader.cpp.o] Error 1

Powerprobot, 2020-03-11:
The same here for SKR1.3...

scorpakascorp, 2020-03-11:
Same situation with SKR Mini E3 1.2. With default "mega2560" config everything ok.

Powerprobot, 2020-03-11:
get the newest commit!  Thinkyhead fixed this at this time!
Work with SKR1.3.

Domush, 2020-03-13:
> 
> 
> get the newest commit! Thinkyhead fixed this at this time!
> Work with SKR1.3.

I have the latest commit as of this moment and still failing to compile with a zillion warnings and one error:

`Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:479:14: error: missing binary operator before token "__RREPEAT2"
  479 |     ( DEFER2(__RREPEAT2)()(ADD1(_RPT_I),SUB1(_RPT_N),_RPT_OP,V) ) \
      |              ^~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:432:21: note: in definition of macro '_IF_1'
  432 | #define _IF_1(V...) V _IF_1_ELSE
      |                     ^
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:479:7: note: in expansion of macro 'DEFER2'
  479 |     ( DEFER2(__RREPEAT2)()(ADD1(_RPT_I),SUB1(_RPT_N),_RPT_OP,V) ) \
      |       ^~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:481:22: note: in expansion of macro '_RREPEAT2'
  481 | #define __RREPEAT2() _RREPEAT2
      |                      ^~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/macros.h:421:24: note: in expansion of macro '__RREPEAT2'
  421 | #define EVAL1(V...)    V`

This continues until it runs of out memory (on a 32GB PC)

Powerprobot, 2020-03-13:
use the Trashcan icon to perform a clean build.  Otherwise take a fresh copy from GitHub. Try this, i think this will help. It help me too sometimes.

Domush, 2020-03-13:
> 
> 
> use the Trashcan icon to perform a clean build. Otherwise take a fresh copy from GitHub. Try this, i think this will help. It help me too sometimes.

Which build are you using? I'm using bugfix-2.0.x
I noticed nobody mentioned the branch they were trying to build.

Powerprobot, 2020-03-13:
Yes, i use the same: Bugfix-2.0.x  here the last commit i compiled sucessfull yesterday: [Last working commit on my system](https://github.com/MarlinFirmware/Marlin/tree/1f55986ed2aeb9db30b2438b125aa8ce64a5f09f)

this is 3...4... commits behind the actual commit at this time.

Domush, 2020-03-13:
Thank you @Powerprobot - Grabbed a completely fresh branch of bugfix-2.0.x and it's compiling fine with SKR 1.3.

Now to redo all of my massive list of customizations, but at least I have a working base, now.

ellensp, 2020-03-13:
Current bugfix 2.0.x with provided configuration files also compiles without issues for  mks robin nano.

Powerprobot, 2020-03-13:
I save my configuration and use VSCode to compare two files. so i can see the original file and my cofiguration and make the changes.

closed 2020-03-14 by thinkyhead