MarlinFirmware/Marlin #17128: [BUG] Initial Homing does not work on any axis

tmartin007, 2020-03-11:
**Expected Behaviour:**

Issuing G28X or G28Y or G28Z should always home the respective axis.

**Actual Behaviour**

Issuing G28X or G28Y or G28Z right after power-on always moves the respective axis a short distance into the direction that is opposite to the definded home-position and stops.

Any successive invocation of G28X or G28Y or G28Z behaves as expected (see above).

**Steps to reproduce**

1) install Marlin 2.0.4.4 with the attached configuration files
2) From a connected terminal console invoke either G28X or G28Y or G28Z
3) Observe the **Actual Behaviour**
4) Again invoke either G28X or G28Y or G28Z
5) Observe the **Expected Behaviour** as it is also expected after 2)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4320796/Configuration.zip)

tmartin007, 2020-03-13:
Sharing some additional evidence:

1) 
The effect that the respective axis moves a short distance into the opposite direction (as described above as "actual behaviour") is due to the HOMING_BACKOFF_MM settings (originally set to "5, 5, 5").  - Changing these settings to "0, 0, 0" avoids this effect.

2) 
the problem appears to be dependent on INVERT_X_DIR, INVERT_Y_DIR and INVERT_Z_DIR
Background: I am controlling a CNC6040 with a SKR Pro V1.1. - The origin of the coordinate system shall be on the "bottom"-right (or "front"-right, if you will). Also, the zero position of the z-axis is all the way "UP".
Therefore, INVERT_X_DIR, and INVERT_Z_DIR are set to "true". Further testing reveals, the Y-axis is not affected by the problem. - Only when setting INVERT_Y_DIR to t"true" as well, the y-axis homing behaves as _"incorrectly"_ as the other x- and z-axis do.

I.e: it takes repeated (e.g.) "G28 Z" to physically home the (e.g.) Z-axis.

Updated **Actual Behaviour**:

Issuing G28X or G28Y or G28Z right after power-on, always moves the respective axis a very short distance towards the home-position and stops. - Marlin, however, considers this "homing" successful - as if the X/Y/Z_STALL_SENSITIVITY is too high. (Reducing the sensitivity has no effect, though.)

It takes several (usually 2 or 3) invocations of (e.g.) "G28 Z" to actually move the axis to it's home position.

Once the homing eventually becomes effective (as in "_moving the axis to the home postition_") any successive invocation of G28X or G28Y or G28Z - behaves correctly.

Updated **Steps to reproduce**

    0. observe that INVERT_X_DIR and INVERT_Z_DIR are set to "true"

ellensp, 2020-03-13:
origin ""front"-right,"
There are only two valid origins in marlin, front left, or back right.   I suspect this is part of your issue.

tmartin007, 2020-03-13:
@ellensp: Thank you very much for your hint! - Can this explain, why everything perfectly works, though, once the homing has actually happened (which it eventually **always** does after 2 or 3 of tries) ?

Geitde, 2020-04-03:
Not sure if this is related, but I had to add a delay after homing an axis sensorless with TMC2209. Without the delay it looked like Marlin is skipped homing for the second axis and only performs the back off movement.

1. Homing Y
2. back off Y 10mm
3. (missing in 1/5 cases) Homing X
4. back off X 10mm
 
In 1/5 cases homing works fine and step 3 is done like it should.

I fixed this behaviour by adding a delay after the Y homing and its back off. Now homing X perfectly works, too. And more important: It works every time.

I am sure if I would use sensorless homing for Z, then I would need an additional delay between X homing and Z homing, too.

boelle, 2020-06-20:
@Geitde is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle