MarlinFirmware/Marlin #17130: [BUG] Sensorless Homing Issue

jgoo9410, 2020-03-11:
Issue:
The motors turn turn fine when jogged and commanded (G1). When homed, motor makes whining noise, jerks left and then right and monitor reports homing successful. Looks to me like the motors are being driven incorrectly which is throwing DIAG_1 LOW and marlin believes it has homed correctly. I have checked that the homing federates are consistent with the default federates etc. All end stops reporting open when idle.

Diag 1 - PA0 (22 in Marlin) Pullups are enabled and end stops are inverted.

```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 30, 30, 4000, 500 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_MAX_FEEDRATE          { 333, 333, 5, 25 }

//#define LIMITED_MAX_FR_EDITING        // Limit edit via M203 or LCD to DEFAULT_MAX_FEEDRATE * 2
#if ENABLED(LIMITED_MAX_FR_EDITING)
  #define MAX_FEEDRATE_EDIT_VALUES    { 600, 600, 10, 50 } // ...or, set your own edit limits
#endif

#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 100, 10000 }

//#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
#if ENABLED(LIMITED_MAX_ACCEL_EDITING)
  #define MAX_ACCEL_EDIT_VALUES       { 6000, 6000, 200, 20000 } // ...or, set your own edit limits
#endif

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves

#if AXIS_IS_TMC(Y). //x is the same
    #define Y_CURRENT       800
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
  #endif

  #define SENSORLESS_HOMING // StallGuard capable drivers only

  /**
   * Use StallGuard2 to probe the bed with the nozzle.
   *
   * CAUTION: This could cause damage to machines that use a lead screw or threaded rod
   *          to move the Z axis. Take extreme care when attempting to enable this feature.
   */
  //#define SENSORLESS_PROBING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  8
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  8
    //#define Z_STALL_SENSITIVITY  8
    #define SPI_ENDSTOPS              // TMC2130 only // have tried with this disabled
    #define IMPROVE_HOMING_RELIABILITY
  #endif
```

Any suggestions would be appreciated.

alieneila, 2020-03-12:
Raise the sensitivity to around 60 and it should work.

jgoo9410, 2020-03-12:
Unfortunately that is not the issue, I have tried lowering the sensitivity already. It is as if the driver is entering spread cycle during a homing move. How could that happen?

hliebscher, 2020-03-12:
witch tmc driver?

fab4fun, 2020-03-12:
It's has to disable stealthchop while homing because stallguard only works in spreadcycle.

zech, 2020-03-13:
@fab4fun On TMC2209 StallGuard4 is for StealthChop mode.

> 11 StallGuard4 Load Measurement
StallGuard4 provides an accurate measurement of the load on the motor. It is developed for operation
in conjunction with StealthChop
[Source Site 55](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf)

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle