MarlinFirmware/Marlin #17134: Power Supply Control feature

Ski3d, 2020-03-12:
Hello!,

I would like to see if i could get an auto shut off feature like below, I use BTT Relay V1.2 on my 3d printer and it controls the mains power to the power supply and also power to the SSR. So in my case if my SSR got stuck in the on position, Marlin could cut power to the whole printer.

    Power Supply Control

//#define AUTO_POWER_OFF_BED_TEMP 130 // (°C) Turn off PSU if over this temperature
//#define AUTO_POWER_OFF_HOTEND_TEMP 300 // (°C) Turn off PSU if over this temperature
[image]

Or

Another idea if we could add a pin to Thermal Runaway Protection to do the same thing, as if a mosfet got stuck on.

#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
#define THERMAL_PROTECTION_BED // Enable thermal protection for the heated bed
#define THERMAL_PROTECTION_CHAMBER // Enable thermal protection for the heated chamber

mvcelicka, 2020-10-19:
I would like to vote for this. In comunity discussion I can see more an more problems with mosfet BTT SKR v2 - mosfet is burned, printer is reaching max temp configured in firmware, but cannot switch of heating as control device is dead - very dangerous. With mentioned relay, this can be easily solved, if printer can send PSU off signal when thermal runaway or max temperature is reached. Also writing some error log to SD card would be nice (with error handled by shutdown).

ManuelMcLure, 2020-10-19:
This is already available. If you enable `PSU_CONTROL` whatever pin is defined as `PS_ON_PIN` in the will be controlled by Marlin. If there is a thermal runaway the `kill()` function will turn this pin off. Also, even without `PSU_CONTROL` enabled you can define `SUICIDE_PIN` as yet another pin that will be triggered in case of `kill()` being called.
These two pins plus the existing thermal runaway protection will provide what you are looking for.

mvcelicka, 2020-10-19:
> This is already available. If you enable `PSU_CONTROL` whatever pin is defined as `PS_ON_PIN` in the will be controlled by Marlin. If there is a thermal runaway the `kill()` function will turn this [...]

Thanks for this info. I was searching some Marlin documentation and was not able to find info about. I will test it. 
Thanks again!

slowbro, 2021-07-03:
Indeed, `kill()` calls `minkill()` which has this line:

```cpp
  TERN_(PSU_CONTROL, PSU_OFF());
```

So, this is implemented via thermal runaway protection.

closed 2021-07-03 by thisiskeithb