MarlinFirmware/Marlin #17136: [BUG] Compiler error SERIAL_ECHOLNPAIR(MSG_FREE_MEMORY, freeMemory(), MSG_PLANNER_BUFFER_BYTES

angelux3d, 2020-03-12:
### Bug Description
compilation error in MARLIN 2
Some help Please.?

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
1 download Marlin2.0
2. [Second Step]
downloa Marlin 2bugs 20..x
3. [and so on...]
Altered confg.h and throw this error.
**Expected behavior:** [What you expect to happen]
0 errors
**Actual behavior:** [What actually happens]

#### Additional Information
[image]

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-12:
Without config files, and educated guess is you changed serial port to something not possible on your controller. NB. This is the serial port on the controller, not the related to the serial port on the pc.

angelux3d, 2020-03-12:
OK thank you i am stucked here 3 days and dont know what to do.
[config angel.zip](https://github.com/MarlinFirmware/Marlin/files/4325272/config.angel.zip)
These are, i dident touch, advanced file.
Thanks

ellensp, 2020-03-12:
Provided configuration files compiles fine on platformio.

ellensp, 2020-03-12:
What version of arduino IDE are you using?

thinkyhead, 2020-03-14:
At this time it is recommended that you use https://marlinfw.org/docs/basics/auto_build_marlin.html with Visual Studio Code to build and upload Marlin 2.0.x.

closed 2020-04-29 by thinkyhead