MarlinFirmware/Marlin #17138: [BUG] (short description)

ElTel15, 2020-03-12:
It has been suggested that my problem may be a bug.  I have tried to find solutions on several reprap and kiwi forums as well as other places and found no solution.

Firmware - Marlin Bugfix2.0.x
1 connect pronterface
2  load sliced gcode file (Cablechainx4.gcode)
3  send to rpinter.

**Expected outcome:**
Satisfactory print

**Actual outcome:**
Printer attemts to begin the print at 7.25 mm (approx, measured by feeler gauges) above print bed.

I have configured Marlin to set the noxxle just above the heat bed and it homes to that position.  I sliced my model (Drag chain links) in Slic3r and then used Pronterface to control my self built Mendel / i3 printer.

I have previously printed this model when I first built the printer some years ago when bed levelling was done manually with microswitch min_endstops.  I have upgraded machine after period on non use with Marlin Bugfix2.0.x and ABL with an inductive probe.  The machine homes properly but when I start to print the nozzle hovers above the bed by about 7.25 mm and attempts to start print.  The Slic3r gcode directs the z axis to 0.3 in it's initial G1 code.  I am completely at a loss as to how to explain this behaviour and would be grateful for any help.  It has been suggested that it is a possible Bug in Marlin.

I am unable to attach my config.h or my model gcode or stl files where should I do that?

[Slic3r gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4323633/Slic3r.gcode.txt)

ellensp, 2020-03-12:
You just drag the file on to the page. but it only accepts zip or text files, so it is best to put all changed files in a zip archive and attach that

ElTel15, 2020-03-12:
Hi ellensp
Thanks for the adviceI have now attached a zip file containning my config.h and the model file gcode as sliced in Slic3r i.e. CableChainx4.gcode.  If someone could test them I would be grateful,  If not a bug I can look elsewhere

best regards

TerryD

[Files.zip](https://github.com/MarlinFirmware/Marlin/files/4323961/Files.zip)

ellensp, 2020-03-12:
Inductive probes...  
I believe the issue is #define NOZZLE_TO_PROBE_OFFSET { 58, -10, -0.9}
The Z point is where the probe triggers in relation to the nozzle, not the height above the nozzle of the physical end of the probe.
you say its 7.25 high so give { 58, -10, 6.35} a go.  (7.25 - 0.9)

ElTel15, 2020-03-12:
I will try that tomorrow, I have to take my cat for chemotherapy now.

ElTel15, 2020-03-16:
Hi, This is still active but I am unable to test my machine until Monday 16th March as I have broken the glass bed and unable to replace until the local glass supplier opens.

closed 2020-06-20 by boelle