MarlinFirmware/Marlin #17139: [PROBABLE] [BUG] SKR 1.4 TURBO not home in center [stallguard] tmc2209 bltouch

zonalimitatore, 2020-03-12:
### Bug Description
With a Ender3 with Skr 1.4 Turbo motherboard

_When do a G28 axis X & Y go at home,but after that, Z probe in 0 position and not in center of the bed
_From config,should go home first Y and then X, instead they both leave together 

I've already cutout the diag pin for X and Y axes

Here what's up....Yeah sensitivity was too lower.
[Attachment_1583942142.zip](https://github.com/MarlinFirmware/Marlin/files/4323731/Attachment_1583942142.zip)

### My Configurations

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/4323698/conf.zip)

### Steps to Reproduce

**Expected behavior:** [An Y and after X homing,in center of the bed]

**Actual behavior:** [Y and X home togheter. Z probing at X and Y 0]

zmiguel, 2020-03-14:
Did you Load defaults and stored them to eeprom after installing marlin? 

M502 followed by M500 to clear old settings in eeprom.

zonalimitatore, 2020-03-14:
Yes. I do that every time

boelle, 2020-06-20:
@zonalimitatore is the issue gone?

zonalimitatore, 2020-06-20:
I've do that job for an user,but I've never receive a response after tell him to do some mod to config and conditional files. I think now work,but I can't be sure ☹️

boelle, 2020-06-20:
i will close then, we can always reopen

closed 2020-06-20 by boelle