MarlinFirmware/Marlin #17140: Delta kinematics inverted mounted [FR] (feature request title)

mac7988, 2020-03-12:
How can I calculate arm lenght for 3 axis if the delta was mounted like image below?
[image]

Traditionally the delta configuration in on 3 towers. 
[image]

ellensp, 2020-03-12:
Surly the length is the same. From center of each joint. But your going to need a new kinematic model for that type of machine.

mac7988, 2020-03-12:
> d

Is this something that is possible to do?

ellensp, 2020-03-13:
Possible yes... have the resources and the interest to do it.. unlikely.

pablopeu, 2020-03-14:
The kinematic model is different because you now have angles and the definitions for a delta machine don't

sjasonsmith, 2020-09-11:
@mac7988 is this request for something which actually exists, or is this just a hypothetical question?

It is already hard enough to find people willing to support Delta kinematics. It is unlikely anybody would ever volunteer to implement this without being directly involved in the project to create the machine. In other words, if this is something you really want, you will likely have to do it yourself.

sjasonsmith, 2020-11-22:
Given the lack of response to my earlier comment, I am going to close this FR. While this is an interesting thought experiment, I don't see any way this would ever be implemented unless it were contributed by the person creating the printer. There is also no information provided regarding _why_ anyone would use such as design.

Adding new kinematics because they are interesting without practical implementations complicates maintenance without providing any value.

closed 2020-11-22 by sjasonsmith