MarlinFirmware/Marlin #17144: [BUG] MINTEMP error on Monoprice Maker Select V2 (Wanhao Duplicator i3)

rgov, 2020-03-12:
### Bug Description

When I turn on the printer, the temperatures of the hot end and bed register as 5°C. The actual temperature of the bed, as measured by an infrared thermometer, is about 18°C. It is cool but not cold.

When I try to preheat the printer for PLA, it errors with a `MINTEMP` error and asks to be reset.

One of the temperature readings did increase from 4° to 5° but has not increased further.

I have previously done a few prints but I have not had Marlin installed for long. I did not encounter this issue with the stock firmware.

### My Configurations

- Monoprice Maker Select V2 aka Wanhao Duplicator i3
- Marlin 2.0.4.4
- [Configuration files from here](https://github.com/MarlinFirmware/Configurations/tree/master/config/examples/Wanhao/Duplicator%20i3%202.1)

Note that the README for the configuration tells me to copy `thermistor_99.h` and `thermistors.h` but this doesn't seem to be needed anymore.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4326936/Configuration.zip)

rgov, 2020-03-12:
Don't know what I'm doing really, but:

```
echo:; PID settings:
echo:  M301 P29.00 I2.00 D97.00
```

rgov, 2020-03-12:
I used a hair dryer on the printer for a few minutes to heat it up to a reported temperature of ~12°C for both the bed and hot end. Then I was able to preheat the PLA.

ellensp, 2020-03-13:
Take a look at your melzi. Locate R4 and R5 
[image]
If the number on the resistor reads 1002 its a 10k pullup and you use thermistor type 99
other wise if it says 472 you have to select correct thermistor with a 4.7k pullup, probably a type 11

rgov, 2020-03-15:
Thanks @ellensp. I did inspect the Melzi and found that R4 and R5 were labeled 472. How would I know that type 11 ("100k beta 3950 1% thermistor (4.7k pullup)") is the correct choice here?

boelle, 2020-06-20:
@rgov is the issue gone?

rgov, 2020-06-20:
Yes, I think this is resolved by following @ellensp's great advice above. It would be interesting to know how it was determined. I've also shared this thread with other people using the same printer.

closed 2020-06-20 by rgov