MarlinFirmware/Marlin #17146: [BUG] Junction Deviation is Borked....

DaMadOne, 2020-03-13:
There are numerous reports of JD affecting prints when LA is used. There has been much discussion about it and it seems the bug reports get closed due to inactivity most of the time. But something isn't right,

From left to right is Cura with JD enabled, then classic jerk. The 3rd and 4th print is Prusa Slicer both with JD enabled and without. I don't use Cura much and as such my profile is probably not that dialed in an why it doesn;t look as good..

I'm only showing it to show that the slicer used doesn't really matter much. It doesn't take a rocket scientist to see the prints with classic jerk are better.The 1st and 3rd print are jagged AF and the 2nd and 4th ones are much better.

With JD enabled the extruder goes nuts on circular prints and is very herky jerky producing the 1st and 3rd print of the Tux model that is not right.. . Without JD enabled circular prints are very smooth. Logic dictates that JD should be much better that classic jerk... but real world tests show otherwise.

I'm running an Ender 3 Pro with an SKR E3 DIP with TMC2209's. 

Click [HERE](https://i.imgur.com/p3AnQeY.jpg) for comparison of what is happening. I printed them with the same Marlin 2.0.4.4 firmware so configuration doesn't really matter but if you want me to post my config files then let me know.

thisiskeithb, 2020-03-13:
> I printed them with the same Marlin 2.0.4.4 firmware so configuration doesn't really matter but if you want me to post my config files then let me know.

Yes, please zip up your configs and attach them for review.

DaMadOne, 2020-03-13:
Don't take my report as "talking shit"... That's not my intention at all. I hope you take a look at my configs and tell me I'm and idiot... I've spent the last couple of days making tweaks to various things and getting nowhere. For the most part my prints come out one of two ways.. looking like Tux 1&3 with JD enabled or Tux 2&4 with JD disabled.

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/4327751/Marlin_config.zip)

DaMadOne, 2020-03-13:
What's weird is that it doesn't affect all circular  movements.  So to me.. with my limited knowledge it seems like some sort of math issue. This particular Tux model is just the worst I've come across.

qwewer0, 2020-03-13:
Printed the bottom of the model, and the extruder is indeed goes crazy. [Here](https://imgur.com/a/OZh930y) are my results.

Latest Bugfix, Ender 3, SKR Mini E3 v1.2

thisiskeithb, 2020-03-13:
@DaMadOne: You've overridden the default `MINIMUM_STEPPER_POST_DIR_DELAY`, `MINIMUM_STEPPER_PRE_DIR_DELAY`, and `MINIMUM_STEPPER_PULSE` settings. Those are set for you automatically, so were defaults (commented out) not working on your TMC2209s?

qwewer0, 2020-03-13:
@thisiskeithb It is all disabled for me. I know that you didn't wrote it to me, just wanted to say it.

thinkyhead, 2020-03-14:
It may help to set the slicer to use minimum length segments. I'm not sure how a lot of short segments interact with these various settings versus curves with longer segments. This would be a good area of exploration.

DaMadOne, 2020-03-14:
> @DaMadOne: You've overridden the default `MINIMUM_STEPPER_POST_DIR_DELAY`, `MINIMUM_STEPPER_PRE_DIR_DELAY`, `MINIMUM_STEPPER_PULSE`, and `MAXIMUM_STEPPER_RATE` settings.
> 
> Those are set for you automatically, so were defaults (commented out) not working on your TMC2209s?

That change was made after I was already having problems. Another bug report here that was already closed but had a similar sounding issue mentioned those settings. So I looked at it and tried it. The comments pertaining to those config lines seem to suggest the settings I applied are the "best default" settings for the stepper drivers I'm using so I figured it was worth a shot. I didn't notice anything negative from them in my testing so I left them. That same bug report did show that they used settings in the middle of all the defaults(and different from mine) and got positive results from it.. I have not tried those exact settings because the settings they said they used seemed to almost match the numbers in the commented out lines that I enabled and changed but saw no difference In my prints. So it didn't seem worth while to spend anymore time changing those.

I couldn't find it last night but another report had some videos posted in the comments. If you watch the before and after videos and try to filter out the other noise and just listen to the extruder then you can hear the difference. That sound it exactly what I'm seeing/hearing. 

Another model it show up on consistently is the benchy. Most of the boat hull prints well. There is a series of layers that cause this issue though. It's nowhere near as extreme as the tux model but you can't miss it when comparing LA with and without JD. Maybe someone who understands the math behind LA and JD can take a look at those models maybe have an idea of whats going on.

DaMadOne, 2020-03-14:
> It may help to set the slicer to use minimum length segments. I'm not sure how a lot of short segments interact with these various settings versus curves with longer segments. This would be a good area of exploration.

> It may help to set the slicer to use minimum length segments. I'm not sure how a lot of short segments interact with these various settings versus curves with longer segments. This would be a good area of exploration.

I will take a look at this tomorrow and will let you know the results.

Moyster, 2020-03-14:
> It may help to set the slicer to use minimum length segments. I'm not sure how a lot of short segments interact with these various settings versus curves with longer segments. This would be a good area of exploration.

Could very well be, I guess OP should check how the layers are "drawn" in its slicer preview, additionally CNCKitchen has a video/semi-tutorial on youtube about this : https://www.youtube.com/watch?v=Hvw3DrVAeTA  
I found tweaking these settings gave me much smoother circles when openscad models were exported with a lot of definition ($fn >= 128)

qwewer0, 2020-03-14:
In my case there are no pausing at all, just continuous X / Y motor stuttering, that causes the rough layers/surface noticeable on the prints, but I can feel it on the stepper motors too when printing.

Moyster, 2020-03-14:
You didn't post your configuration so it's hard to tell what could be wrong from pics with the light/seams at a different place on each pictures.  

What's your config / setup / JD value ? Bowden ? Direct drive ? Did you do the jerk to JD conversion correctly ? Did you try other values for JD ?  

I have pretty round circles on a bowden/cartesian setup with JD (0.12) / S-curve on, no LA, and :  
maximum resolution : 0.5  
max travel resolution : 0.8  
max deviation : 0.08  

print speeds from 40 to 60-65-ish, 0.1 to 0.3mm on a 0.4mm nozzle.

qwewer0, 2020-03-14:
Sorry, didn't thought that my confings too are needed, sorry.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4332887/Marlin.zip)
- Direct drive
- LA = 0.17
- Acc = 500
- JD = 0.08 (0.4*10*10/500)
- I tried higher (18), lower (0.02) and in between a few

Moyster, 2020-03-14:
Well, I'm not saying there is no bugs, but both you and OP has that issue with JD, so it's best to check / cross-check both of your configurations, make sure there's a legit need for marlin devs to dig deeper.  

Might be a bit of a blind shot, but did you try to use "a well known 3D printer model's config" regarding JD / accel ?  
Anet A8 examples config uses JD 0.10 and 400 accel, and it has a direct-drive and a wobbly frame, so I'd assume these values are "conservative".

qwewer0, 2020-03-14:
I "followed" the Ender 3 and BTT SKR Mini E3 v1.2 example files, and I made my own [guide](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/).
I get a lot of questions day to day, so I rechecked my configs now a dozen-dozen times.

The annoying part is that with Classic Jerk I get better result in that model, and I can feel the steppers too, that they are not stutter when printing the surves.

swilkens, 2020-03-14:
A different avenue; are you printing from SD / Storage media or are you printing over the serial port?

qwewer0, 2020-03-14:
I print from SD card.
Don't know if it helps, but my baudrate is 115200 ([page 799](https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf), 72Mhz)

DaMadOne, 2020-03-15:
I'm getting down to further testing today. I will use this post as a sort of scratch pad that I will update periodically with what I am trying and the results. 

First up today was to upgrade to 2.0.5. [HERE](https://github.com/MarlinFirmware/Marlin/files/4334592/Malin_config.zip) are the config files as they sit right now. I cut out and am only using part of the middle of the tux model to speed up test prints. First print shows the stutter.

Next test was to try printing the same gcode from SD since I've been using Octoprint. Stutter is still there so that should rule that out.

I've decided to try to slow things down to see if speed makes any difference. Will report back...

DaMadOne, 2020-03-15:
*moved this from my other comment*

Question for someone in the know. Kinda *thinking* out loud. Is the math for JD more/less/sameish to the math for CJ(classic jerk) when it comes to CPU usage? Wondering if it's possible to be hitting processing limits of the STM32's on these boards? Probably not? Since the stutter to me seems more like a ton of retracts than it does some processing problem. When the extruder is stuttering it doesn't seem like X/Y are also stuttering but maybe it's just harder to see and hear.

qwewer0, 2020-03-17:
Printed another [example](https://www.thingiverse.com/thing:3469190) piece, JD top, CJ bottom:
[Mine](https://imgur.com/a/YZoJJon)
[Helper](https://imgur.com/sf9Lmhh)

Moyster, 2020-03-17:
@DaMadOne : what thinkyhead was referring to (minimum segment length) is what the video I linked talks about.  
You should try a print with exaggerated resolution settings, see if at 50mm/s it still struggles, it could be a performance issue, or something wrong with "something something PWM pulse" when there's a lot of tiny segments.  

https://github.com/MarlinFirmware/Marlin/pull/17171  
This could maybe improve the situation ? Using a larger buffer size (64) and increasing slowdown factor to 4.

ManuelMcLure, 2020-03-17:
> Based on the 2nd one above with JD enabled it's not possible to tune EJERK from the menus or gcode anymore as far as I can tell. M503 does not show an M205 value for E (M205 B20000.00 S0.00 T0.00 J0.08) and it's not in the jerk menus on LCD. Or am I missing something? Is it even worth trying to mess with this value based on the issue I'm having?

This is correct - if you're using JD the extruder jerk is calculated according to the following formula:
```
#define GET_MAX_E_JERK(N) SQRT(SQRT(0.5) * junction_deviation_mm * (N) * RECIPROCAL(1.0 - SQRT(0.5)))
```
where ``N`` is the maximum acceleration on the extruder axis.

qwewer0, 2020-03-17:
@Moyster I changed `BLOCK_BUFFER_SIZE` to 64, and `SLOWDOWN_DIVISOR` to 4, but it [did not helped](https://imgur.com/a/KrMSd2Z).

psavva, 2020-03-17:
I see you have Linear Advance, and working with Cura...  That will not work very well.  Reslice with Slicer using Lin advance, or alternatively, turn off linear advance and see if you can simulate the issue.  There is an incompatibility between Curas extrusions widths and Lin Advance...

qwewer0, 2020-03-17:
> There is an incompatibility between Curas extrusions widths and Lin Advance

Don't know how is that even makes sense, as I printed it with and without LA, and it was the same, but ok.
I resliced the [model](https://www.thingiverse.com/thing:1809605) with prusaslicer, and printed it with JD and CJ, and to my surprise the two came out the same. Both of it was better than the cura file with JD, but worse than the cura with CJ, but I think it is just my setting difference.
So, I don't know how or why, but it might be a problem only with cura and not with JD in general, and I hope others can test it too, to confirm/refute it.

DaMadOne, 2020-03-18:
> I see you have Linear Advance, and working with Cura... That will not work very well. Reslice with Slicer using Lin advance, or alternatively, turn off linear advance and see if you can simulate the issue. There is an incompatibility between Curas extrusions widths and Lin Advance...

This isn't remotely true. Is it true that Cura makes MANY assumptions? Yes it does. Does Cura break LA? then the answer is no. You can tell by reading the OP and checking out pics that I posted that the slicer doesn't matter. I can print a terribly calibrated Cura prints with LA and JD enabled and tell the difference. It's only slightly different from my better calibrated Prusa Slicer profile between JD enabled and not. It's easy enough to turn off most of Curas advanced features.

DaMadOne, 2020-03-18:
On to block buffer size. I didn't mention it but things I tried didn't make a difference.

DaMadOne, 2020-03-18:
The more I dive down this rabbit hole of bug reports and Pull Requests the more I'm convinced this isn't an immediately solvable issue. LA just wasn't developed with bowden extruders in mind. So I can either jump to a direct drive system or disable LA.

DaMadOne, 2020-03-18:
@ManuelMcLure I'm a little inebriated right now but I'l check out the math tomorrow. Would be interesting to see what my jerks value end up at once running through that math,

psavva, 2020-03-18:
> > I see you have Linear Advance, and working with Cura... That will not work very well. Reslice with Slicer using Lin advance, or alternatively, turn off linear advance and see if you can simulate the issue. There is an incompatibility between Curas extrusions widths and Lin Advance...
> 
> This isn't remotely true. Is it true that Cura makes MANY assumptions? Yes it does. Does Cura break LA? then the answer is no. You can tell by reading the OP and checking out pics that I posted that the slicer doesn't matter. I can print a terribly calibrated Cura prints with LA and JD enabled and tell the difference. It's only slightly different from my better calibrated Prusa Slicer profile between JD enabled and not. It's easy enough to turn off most of Curas advanced features.
> 
> It doesn't help to regurgitate the crap people read and hear from reddit.

Please have a look at the issue reported 3 years ago https://github.com/MarlinFirmware/Marlin/pull/5928#issuecomment-283113646

qwewer0, 2020-03-18:
> Please have a look at the issue reported 3 years ago [#5928 (comment)](https://github.com/MarlinFirmware/Marlin/pull/5928#issuecomment-283113646)

I read it through, and it is very interesting, but it doesn't explains why did I get the same bad JD result with or without LA enabled in the firmware. Is there some other advance marlin feature that is also incompatible with cura and needs to be turned off in order to test JD?

psavva, 2020-03-18:
Not that I'm aware of

qwewer0, 2020-03-18:
I sliced and printed [this model](https://www.thingiverse.com/thing:1809605)'s bottom 5mm with Cure, PrusaSlicer, and Fusion 360, and printed them with the latest bugfix, with JD and with CJ.
The results were as expected - at least for me - as all of it came out worse with JD than CJ.
It may not appear well in the [pictures](https://imgur.com/a/g8VUGbh), but in reality it can be seen well and can be feel its surface.
- The two Cura prints had the biggest difference, as the one with JD is the worse of all, but the one with CJ is the second best.
- PrusaSlicer is in between the two Cura prints, as the CJ is only helped just a little, but the JD one is still turned out worse.
- Fusion 360 with JD yielded a better result than the two PrusaSlicer print, but with CJ it become the best of all.

qwewer0, 2020-03-19:
I think this should be tested by others that don't have an SKR board, to know that this is or isn't a hardware issue.

Seref, 2020-03-21:
Hey there, I'm also using a SKR Mini E3 1.2 board and also got weird artifacts when enabling JD 
[image]

The lines are always wavy around curves, doesn't matter if I enable or disable linear advance it's always a bit wavy.
I don't get them when switching to classic jerk and enabling linear advance, which is weird.
(I got a custom printer, though it's similar to an Ender 3 and is also a Bowden style 3D Printer).

qwewer0, 2020-03-21:
If this is an issue specifically with the SKR Mini E3 v1.2 and maybe with other SKR boards too, then how can we make sure of it, then solve it?
I'm open to ideas.

swilkens, 2020-03-21:
> If this is an issue specifically with the SKR Mini E3 v1.2 and maybe with other SKR boards too, then how can we make sure of it, then solve it?
> I'm open to ideas.

Build back your original board, use the same firmware options and verify on the original board?

qwewer0, 2020-03-21:
> Build back your original board, use the same firmware options and verify on the original board with the same gcode?

I would have done that already, but I no longer have the original board.

BarsMonster, 2020-03-22:
Just want to leave a link to potentially relevant issue: #15473 

What I learned there is that JD with LA and/or acceleration control in Cura makes an easily reproducible issue, but it is present even without LA/JD/acceleration control, though very rare (like once per 200 layers).

You can try to enable acceleration control in Cura (assign different accelerations for infill/perimeters/e.t.c) and see how prints will get completely wrecked. 

On the bright side, currently on SKR PRO 1.1 I see much less (if at all) of an issue. So the issue might be more repdoducible on slower CPUs.

CarlosGS, 2020-03-29:
I think I'm having the same issue as @DaMadOne, tomorrow will try disabling Junction Deviation and see if those weird erratic arcs become smooth.

Marlin version: 2.0.5.2
Printer: Hephestos 2
Electronics: 8 bit
Junction deviation: enabled
Linear advance: disabled
S-curve: enabled

thinkyhead, 2020-03-31:
Duplicate of #17342

richfelker, 2020-03-31:
Just saw this and the linked video:

> Could very well be, I guess OP should check how the layers are "drawn" in its slicer preview, additionally CNCKitchen has a video/semi-tutorial on youtube about this : [...]

Unfortunately this is not a solution. Even Cura's default max resolution/deviation settings are way too high to accurately reproduce fine detail, and due to numerical instability, they produce inconsistent results on consecutive layers, resulting in non-smooth surfaces. For fine detail, max deviation needs to be on the order of microstep size. There's a similar problem in Marlin firmware that compounds on top of this too, `MIN_STEPS_PER_SEGMENT`. See part of my answer at https://3dprinting.stackexchange.com/a/12075/11157 for details.

Any real viable solution of these problems needs to avoid depending on "make gigantic imprecise segments".

thinkyhead, 2020-04-01:
> they produce inconsistent results on consecutive layers

Well this really could be addressed by the slicer authors. I won't list all the potential approaches, but I can think of a few ways to mitigate this effect.

thinkyhead, 2020-04-01:
> Any real viable solution of these problems needs to avoid depending on "make gigantic imprecise segments".

The thing we're trying to look at here is the behavior of junction deviation in the presence of many small segments. In the other thread I was keen to state that we're not trying to solve the issue by this method, but to eliminate factors that could be contributing to the bad acceleration effects. I am asking testers to reduce the number of segments and give feedback so that we can see whether there is any effect and what that effect is. But the deeper investigation involves a lot of logging and doing lots of French curves and Fibonacci spirals.

richfelker, 2020-04-01:
Inconsistency between consecutive layers can be solved by slicers if they care, yes, but loss of detail can't. In any case I'm in full agreement that eliminating other factors (tiny segments) that could be involved with this JD bug is a good debugging technique. I just misunderstood the low-res slicing recommendation as a statement that high-res slicing is undesirable or unsupported in general.

CarlosGS, 2020-04-01:
To clarify, the stuttering appears for **some** arcs.
Even parallel ones, sometimes it stutters and sometimes its perfectly fluid.
So small segments can't be the only cause...

thinkyhead, 2020-04-05:
This is worth a read for anyone using 2208 and 2209… https://github.com/MarlinFirmware/Marlin/issues/11825#issuecomment-421809385

BarsMonster, 2020-04-06:
@thinkyhead I believe stealthchop issues are not relevant as it stops all extrusion permanently, while in this case it misses alot of steps. In my case TMC2209 was in SpreadCycle mode.

qwewer0, 2020-04-06:
> This is worth a read for anyone using 2208 and 2209… [#11825 (comment)](https://github.com/MarlinFirmware/Marlin/issues/11825#issuecomment-421809385)

@thinkyhead So, is it enough to just turn off StealthChop and try to print a sample with JD to test it? Because I did it and it didn't turned out any better, but was loud the whole time...

SKR Mini E3 v1.2, TMC 2209, latest bugfix.

CarlosGS, 2020-04-06:
@thinkyhead My drivers don't support [StealthChop](https://www.youtube.com/watch?v=FqWgTM4di4s) :) they are the DRV8825

XDA-Bam, 2020-04-12:
For anybody looking here, I've posted a [workaround ](https://github.com/MarlinFirmware/Marlin/issues/17342#issuecomment-612664924) in #17342 .

BarsMonster, 2020-04-12:
@XDA-Bam  I've tested MIN_STEPS values of 16 and 6 with JD on SKR PRO. I see hardly any differences in surface quality. MIN_STEPS 16 has slightly "noise" on the surface, but still very far from JERK quality.

Jerk surface quality is better with all MIN_STEPS values (16, 6, 1).

[image]

qwewer0, 2020-04-13:
@BarsMonster Could you share your test file? I would like to test it too, but on SKR Mini E3 v1.2.

BarsMonster, 2020-04-13:
@qwewer0 It's here:
It has very high resolution, and was sliced with "Maximum deviation" of 0.002 in Cura to intentionally produce very tiny segments in g-code. 

[Xtest-HR.zip](https://github.com/MarlinFirmware/Marlin/files/4468931/Xtest-HR.zip)

swilkens, 2020-04-13:
Another dataset; I took the STL from @BarsMonster (https://github.com/MarlinFirmware/Marlin/issues/17146#issuecomment-612678251) and printed the same .GCODE file with JD (J=0.08) and CLASSIC_JERK. (X, Y JERK 10)

Results below. Both printed directly from SD card on SKR Mini E3 v1.2 running 2.0.5.3.

I honestly can't see a difference here. Both with `#define MIN_STEPS_PER_SEGMENT 6`

[image]

qwewer0, 2020-04-13:
With @BarsMonster test file, I can't reproduce the the issue with JD. (SKR Mini E3 v1.2)

BarsMonster, 2020-04-15:
@swilkens @qwewer0 Do you have high resolution enabled in your slicer? You must have segments with length of ~0.1-0.5mm in the output to trigger the issue. With Cura you achieve that by setting Maximum Resolution ~0.2mm and maximum deviation of ~0.002mm (as an ultimate example). 

As we see in the code (https://github.com/MarlinFirmware/Marlin/issues/17342#issuecomment-613798026) , with segments of 1mm or longer - issue would not be reproduced.

qwewer0, 2020-04-15:
@BarsMonster In PrusaSlicer the "Slice gap closing radius" (Cracks smaller than 2x gap closing radius are being filled during the triangle mesh slicing) is set to 0.049, so maybe the max resolution would be 0.1.
But the JD_single_arc_test.gcode [works](https://github.com/MarlinFirmware/Marlin/issues/17342#issuecomment-613602582).