MarlinFirmware/Marlin #17148: [BUG] POWER_LOSS_RECOVERY request on every completed print

5a2v0, 2020-03-13:
### Bug Description

I upgraded my Geeetech I3 ProB with marlin 2.0.4.4 version and with this upgrade when i finish 100% a print I got on the display a confirmation to resume or stop the print!
I did a eprom inizialization and then printed another object but I got same result.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4329387/config.zip)

### Steps to Reproduce

1. Using marlin 2.0.4.4 start a print
2. Wait until finish
3. Read on lcd the request for resume the stopped print also if not power loss problem during print

**Expected behavior:** lcd shows 100% print status and time

**Actual behavior:** lcd shows a request for resume the stopped print also if not power loss 
problem during print

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-03-13:
Give the latest `bugfix-2.0.x` branch a try. Several bugs related to SD printing were squashed recently, so it may have been fixed.

thinkyhead, 2020-03-14:
In your testing please enable `DEBUG_POWER_LOSS_RECOVERY`. This will print extra messages to the console for events like deleting the power-loss recovery file. Do a simple G-code file test, such as a file with just a bunch of `M300` and `M118` commands in it. If there are problems with the power-loss file handling then hopefully the extra logging will expose it.

5a2v0, 2020-03-14:
Ok, tested with bugfix version and printed the same gcode that caused problem before on the 2.0.4.4 
Now the printer finished the job and LCD shows "Press a button" message at the end.
After i clicked once the encoder, lcd shows normal status info with 100% and printing time

So can We close the issue right?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle