MarlinFirmware/Marlin #17155: Changing USB VID / PID

towe96, 2020-03-13:
I'm using a single Raspberry Pi with multiple Octoprint instances. To differentiate between the printers, I use the 99-usb.rules file, assigning a name to each combination of USB VID and PID of the printer boards.

I'm now at a point where I use two very similar boards in two of my printers - a SKR Mini E3 v1.2 in my CR-10S, and a SKR E3 DIP in my EZT T1.
With Marlin 2.0.4.4, the USB ID's end up as "1eaf:0004" for both of those boards.
Adjusting any of the "1eaf" or "0004" strings in the source didn't seem to help. Setting the "USB_DEVICE_VENDOR_ID" and "USB_DEVICE_PRODUCT_ID" also didn't change anything.

How can I change those IDs, so that I can differentiate between my two boards?

Clint75, 2020-03-15:
I have never done this my self but have a look at https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-board-configuration

towe96, 2020-03-15:
That didn't work in itself, but it was definitely caused by platformio.
I've ended up changing every instance of "1eaf" to "1234" in the C:\Users\<username>\.platformio structure - and it worked.
Now to figure out which it was.

boelle, 2020-06-20:
@towe96 is the issue gone?

towe96, 2020-06-20:
Not really, I have to get back to this.
I feel like a config option - probably in the platformio.ini file - would be useful here.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle