MarlinFirmware/Marlin #17157: [BUG] unexpected z height increase during print.

zmiguel, 2020-03-14:
### Bug Description

Z-axis keeps going up while printing first layer, after about 10 minutes into a print without a layer change the Z-axis keeps going up unexpectedly. From observing the Z rods it appears that the Z level compensation from bed levelling stops lowering the print head and always moves it up.

### My Configurations

[GitHub Gist](https://gist.github.com/zmiguel/7cb15fa03d6d4bb1218c997ceea63a49)

SKR 1.4 Turbo board
TMC2209 Drivers
BLTouch
Titan Aero extruder

### Steps to Reproduce

1. Auto home
2. Level bed using AUTO_BED_LEVELING_BILINEAR
3. wait...
4. set fade to 10
5. enable bed levelling
6. store settings to EEPROM
7. print something with a long first layer

**Expected behaviour:** Print goes fine and bed levelling works

**Actual behaviour:** After about 10-12 minutes the Z-axis only moves up and not down while compensating for bed level

#### Additional Information

[pictures here
](https://imgur.com/a/9mfCRbG)

thinkyhead, 2020-03-14:
Usually when we see issues like this it's due to electrical noise, wiring issues, or mis-configuration of Trinamic stepper drivers. I suggest you get help with troubleshooting from one of these community resources before we go on a wild bug chase…

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

zmiguel, 2020-03-14:
@thinkyhead I've had this working for 3 months without issues. This only started after 2 recent firmware updates, onde of them about 2 weeks ago and the latest yesterday. It prints fine smaller models like the common zyx calibration cube. However, on larger models that take longer in each layer, this happens. I haven't changed any wiring or changed the environment the printer is in.

Would it be worth changing the ABL mode and testing again?

I'll check with the latest stable branch and see if it still happens and update this thread.

swilkens, 2020-03-14:
If previous versions are known-good to you, it might be beneficial to go back to one of those versions and verify.

makemerush, 2020-03-14:
I've experienced the same issue on my SKR 1.4 since the build two weeks ago (working fine with previous builds)...

My Z axis is powered by a TMC2130 in SPI mode

makemerush, 2020-03-19:
I thought this was fixed but it happened again yesterday night! :(

GibsDev, 2020-03-22:
I have a similar issue (TMC2130 Einsy Rambo) except my Z starts decreases instead of increases. I technically am just nearing the end of building the machine, so I have no experience with prior firmware versions that did not do this.

zmiguel, 2020-04-01:
To give an update on this. I've been trying multiply bugfix versions and I still get this issue. I've recently moved the printer to a different house and the issue still happens.

I've been able to print that same gcode file successfully 2 more times, and have had it crash on me 5 other times.

I believe it might be somewhat related to electronic noise as @thinkyhead mentioned. but I haven't been able to track it yet. My voltage on the outlet is stable from the multiple tests I have done. my skr board seems to be in good shape. I've also tried to swap the TMC driver for a spare I had and Still got the same issue. So I'm getting out of ideas.

If anyone has any recommendations please let me know.

makemerush, 2020-04-01:
I added a fan to my extruder stepper motor for extra cooling which seemed to help, however that might just be a fluke. Purely qualitative observation at this point...

If it doesn’t re-occur, I wonder if the motor heat was triggering the freeze?

zmiguel, 2020-04-05:
After a few more tests it does seem like this only happens on prints with long layer times, the original gcode file that made me create this ticket consists of 2 models, each of those models prints fine individually, but together the first 3 layers take more than 1h each. I've printed the individual models 6 times each and they all printed perfectly. 

So I don't really know what to think of this issue. On one hand it kind of looks like some electronic interference, but it always prints smaller models fine. Guess I'll have to try another bigger model with longer layer times and see if this happens again.

snekonaplane, 2020-04-23:
I had this issue and I solved it (so far as I can tell) by going into configuration.h and adding a value of 10 to ENABLE_LEVELING_FADE_HEIGHT. My theory is that I initialy set the fade height with M420 and erased it from EEPROM with a M502 during other troubleshooting since it was never defined in the firmware. I'm not sure if, or why, the value being blank causes the issue, but so far this has fixed it for me.

closed 2020-06-20 by boelle