MarlinFirmware/Marlin #17158: G29 Not respecting XY Offsets from M206

RFBomb, 2020-03-14:
### Bug Description

I have a previous issue noted [here](https://github.com/MarlinFirmware/Marlin/issues/17100), but that issue has been resolved I believe. So this is a new issue, technically. 

I'm Attempting to run BL-Touch with UBL on my Ender 3. The issue I am having is that the G29 command is ignoring the offsets from the M206 command, causing the probe to not reach the bed in the first position. 

### My Configurations
The configuration files in the linked post above still apply. The only changes made to them are noted here:
```
#define NOZZLE_TO_PROBE_OFFSET { -41.8, -14.4, -0.7 }
// Certain types of probes need to stay away from edges
#define MIN_PROBE_EDGE 0
----
#define MESH_INSET 1              // Set Mesh bounds as an inset region of the bed
#define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
-----
define Z_PROBE_LOW_POINT          0     // Since I have a Z Offset from the microswitch to determine where nozzle sits just below bed -- Mostly to not crash nozzle into bed too hard

```
- Note that I am using a micro switch for Z homing, so `Z_SAFE_HOMING` is disabled.  
- (0,0,0) would normally be with Z just below the bed and Y would have the nozzle off the bed. 
- The M206 offsets bring (0,0,0) to the nozzle touching the corner of the bed. 

### Steps to Reproduce

1. G28                                              ->> Home the printer using XYZ microswitches (not BL-Touch)
2. M206 X0 Y-10 Z-0.97                   ->> Define the home offsets so the nozzle lands on the bed corner.
3. M851 X-41.80 Y-14.40 Z-0.70      ->> Define the BL-Touch NOZZLE_TO_PROBE_OFFSET (or dont since they can be defined in FW)
4. G29 P1                                          ->> Attempt to generate a mesh

**Expected behavior:** [What you expect to happen]

The probe should move to where the nozzle would be at (X,Y) (0,0) and probe the bed height. This (0,0) position should respect the M206 offsets, because prints respect the M206 offsets. 
My M206 offsets are (0,-10), and the probe offsets are (-41.8,-14,4). So the print head should move to (42.8,15.4) with respect to the M206 offsets + Inset Range. 

**Actual behavior:** [What actually happens]

The print head moves to (43 , 5) according to the display. This causes the probe to miss the bed. 
It would appear that the first probe point is defined as the (X_MIN_POS, Y_MIN_POS) position of the nozzle, rather than where (0,0) lies according to the M206 offsets. 

Strange behavior noted if Y_MIN_POS is negative in firmware though (noted below).
Setting the Y offset in the firmware using the Y_MIN_POS option (then having the M206 Y value be 0) will result in eventually probing the correct 0,0 location. Except for some reason, instead of starting at the (0,0) location, it moves to the (0, Y_BED_SIZE / 2) location for the first probe point. Eventually it does hit (0,0) properly though. 

#### Additional Information

Work-around appears to be hard-code the desired Y offset in the firmware under the `Y_MIN_POS` option.

thisiskeithb, 2020-03-14:
https://github.com/MarlinFirmware/Marlin/issues/17007 might offer some insight.

thinkyhead, 2020-03-14:
The `G29` command probes the mesh which is defined in hardware coordinates. The position of the mesh does not change when workspace offsets (`G92`) or home offsets (`M206`) change. You probably want to set your `Y_MIN_POS` to -10 instead of using `M206` to shift the mesh position onto the bed. The `Y_MIN_POS` defines a physical dimension, whereas things like "home offsets" are logical offsets applied to G-code.

closed 2020-03-14 by thinkyhead

RFBomb, 2020-03-14:
I am using Marlin 2.0.1 (according to the startup screen on my printer). 
All of the above results are true for that firmware version. 

I just tested 2.0.4.4 with the 'workaround' enabled:
- Having a negative `Y_MIN_POS -10` caused the probe to perform the first probe at the middle of the Y_Bed_Size position. This occurred on the first probe attempt. 
- I then set `M206 Y-20` out of curiosity. (This for some reason caused it to probe exactly where I expected the nozzle to be at (0,0), but the location was wrong because the offset was now Y-30 due to having -10 in FW and -20 from M206). This was the second probe attempt, immediately after., without using another G28. (SO `G28 -- G29 P1 -- M206 -20 -- G29 P1` )

To test, I also loaded in a firmware with the `Y_MIN_POS` set to 0 (because that is what most people would leave it as). M206 was never saved to the eeprom, so it defaulted back to 0,0,0 after loading firmware.
- This time, my first round of probes were exactly as my original post described, the probe was off of the bed. 
- I then attempted to perform the same test as above, except this time I chose a Y offset of -110 instead of -20.  My test sequence was `M206 -110 -- G28 -- G29 P1 -- G29 P1` 
- First G29 command off the bed again, just like prior to setting the M206 value. 
- Interestingly, the second test once again probed close to where (0,0) would be according to the M206 offsets. It wasn't exact though, the Y was at '17' instead of '-14.4' like it should have been. No idea where the extra few mm came from, I did not see that occur in 2.0.1 firmware (the numbers added up there). 
```
#define MIN_PROBE_EDGE 0
#define MESH_INSET 0              // Set Mesh bounds as an inset region of the bed
 #define GRID_MAX_POINTS_X 3 
#define RESTORE_LEVELING_AFTER_G28 false
```

Finally, no matter what tests I performed, UBL was only probing exactly 1 XY coordinate when using the `G29 P1` command. So it appears that it is broken in 2.0.4.4. It still counts the points for each time it hits the probe, but it no longer moves XY to the next point.

It appears that this bug the issue is referring to not only still exists in the latest version, but is worse due to only hitting a single point, but also because in 2.0.4.4 it DOES go according to the M206 offsets, but only if you try to run a second time. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4332257/Configuration.zip)

RFBomb, 2020-03-14:
@thinkyhead I believe this issue still warrants taking a look at due to my testing on 2.0.4.4. 
The 'workaround' I posted in my original post is the same one you recommended when closing the post, so I am perfectly fine with using that code in my firmware. (Its just confusing unless you realize it). 
However, my findings from 2.0.4.4 are troubling, and I will be reverting back to 2.0.1 due to the issue.

thinkyhead, 2020-03-14:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

RFBomb, 2020-03-14:
@thinkyhead  -- Just tested the UBL building in the bugfix version you linked. The problem still exists, it only probed 1 location. The bugfix version also gave me an 'EEPROM ERROR' that did not occur on 2.0.4.4.

thinkyhead, 2020-03-14:
> The bugfix version also gave me an 'EEPROM ERROR'

This is usually solved with `M502`, `M500`.

thinkyhead, 2020-03-14:
Let's gather more information about this probing problem…

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Vertabreak, 2020-03-15:
M206 does not account for probe offsets its part of the cnc system i don't think it was ever intended to be used this way given how its functioned for the last few years.

RFBomb, 2020-03-15:
I understand M206 is meant to be purely about where the nozzle is.

To me, if M206 is adjusting where the nozzle is located, and the probe is
supposed to determine a Z height for a nozzle point in the bed, I would
think that those XY coordinates should match up. It seems to me that that
is not the case, since  0,0 nozzle is on the bed and the probe is misses
that point altogether, because the probing should take the M206 offsets
into account.

Like I said though, adjusting the Y_Min position in the firmware works
around the issue, since it shifts the Y0 onto the bed for both nozzle and
probe (instead of M206 which just shifts nozzle)

That was fine in the bug fix version. But both 2.04.4 and bug fix versions
only probed 1 XY point until the grid-count was filled.

thinkyhead, 2020-03-15:
I think you're right that when probing, the values that are stored in the mesh for Z should take the Z home offset into account in some manner. You do want 0.0 to mean the nozzle is on the bed (if the `M851` probe offset isn't compensating for the difference).

If you're using the probe for homing, the `M851` offset doesn't need any change, obviously. But if the method of homing Z and probing the mesh are different then `M851` is one place where the correction can be made.

In olden days the `M206` offset was not a persistent value that we included in the workspace offset, but just an offset that was added to the current position after `G28` and then never otherwise considered. I liked that scheme and argued to keep it, but other considerations superseded. Under that method the mesh wouldn't need any compensation.

Now the `M206` value applies on every `G0` command, so coordinates are shifted on input, and a move to Z=0 will be converted into a "native machine space" move to Z=0.5 or whatever the offset is. There are good reasons to keep it that way for `G92`/`G58` workspaces. I'm not sure it's important to retain that behavior for `M206` home offsets.

So a few questions arise…
- Can the workspace offset aspect of `M206` be preserved (if necessary) while also applying `M206` correction to mesh Z values and / or probe Z values?
- Do we need to have XYZ `M206` home offsets all apply on `G28` (as they did in 1.0.x) or do we only do this with Z?
- What are the current use cases for `M206` that require one approach versus the other?
- How will `M206` changes affect the way `M206` offsets apply at runtime / load / reset?

RFBomb, 2020-03-16:
As far as M206 Z offset is concerned, I think that works fine as is. I
don’t think that the Z probe should take the Z offset into account when
building a mesh, as the mesh is a height map for where the nozzle is at a
given XY point.

Currently, using M206 offset to move Z shifts the entire mesh up or down,
which can be useful to easily shift the entire mesh if without having to
perform M851 Z then G29 again. (Atleast in 2.0.1 firmware).

My arguments were about the XY offsets. Where nozzle XY 0,0 becomes
different than the probed point, due to G29 not taking the XY m206 offsets
into account, which is what I’m arguing for. Note that I’m not sure if the
M206 offset for XY shifts the whole mesh like an M206 Z does.

I believe that M851 should take the M206 XY offsets Into account when
checking positions, that way M851 remains true to definition. (The probe
offsets to nozzle). You could compensate currently by adding your M206
offset to the M851 offset, but that gets convoluted and strays from the
M851 definition.
M851 already takes the Z offset into account, as I note in the response to
your question below:

>> “ In olden days the M206 offset was not a persistent value that we
included in the workspace offset, but just an offset that was added to the
current position after G28 and then never otherwise considered”

 Isn’t this was G92 currently does?

>> “ But if the method of homing Z and probing the mesh are different then
M851 is one place where the correction can be made.”

I agree, and have recently figured (painfully) out how to do so. I’ve also
done a write up on it since I haven’t found a guide written by anyone else
in the matter.
https://reddit.com/r/3Dprinting/comments/fitw4a/guide_for_using_a_z_microswitch_leveling_probe_at/

As far as your questions:

>> “Do we need to have XYZ M206 home offsets all apply on G28 (as they did
in 1.0.x) or do we only do this with Z?”

Don’t they already do this? Having an X offset of -5 set by M206 causes X
to read -5 instead of 0 after a G28. Even if the M206 value was set earlier
than the G28 (such as stores in EEPROM).

>> “What are the current use cases for M206 that require one approach
versus the other?”

I thought M206 was to programmatically set offsets from the home position
to shift nozzle onto the bed instead of forcing it via firmware, which is
where my side of the discussion is based from.  I believed that if you
needed to offset your workspace, G92 was the ideal usage, as G92 is not
persistent in the way M206 is.

I was under impression that M206 should be used only for initial setup due
to its persistence.

>> “Can the workspace offset aspect of M206 be preserved (if necessary)
while also applying M206 correction to mesh Z values and / or probe Z
values?”

Like I note above, I thought this is what G92 was supposed to be for.

Technically M206 does shift the workspace, then G92 shifts that shifted
workspace. Calling M206 after G92 shifts the whole G92 workspace. But that
can be accomplished with another g92 if needed, so G92 is t be better
method for workspace shift I think. Atleast that’s my understanding.
——

Side note: how *does* G92/M206 affect the mesh?
I know M206 Z offset will raise of lower the whole thing. Does a new M206
XY command shift a previously saved mesh?

If you are at X10,Y10 on your mesh, then perform a G92 to make the printer
think it’s at X50,Y50, does it move to the Z location for 50,50?

——

My concern if we do change how these interact is making sure the docs
update to reflect it, and hopefully it won’t break people’s configurations.

thinkyhead, 2020-03-16:
`M206` and `G92` are illusions. Everything in Marlin internally is done in "native machine space." So when you send Marlin a `G0` command the combined `M206` and `G92` offsets are subtracted from the XYZ coordinates right at that moment, and then never looked at again. (Marlin also converts the input to mm, if you're in inches mode.)

Except, when you ask Marlin "what is the current position" it adds back the current `M206` and `G92` offsets to the "native" XYZ position before it shows you the "logical" XYZ position. (It also converts the output to inches, if you're in that mode.)

Outside of these two points (XYZ input and XYZ report), `G92` and `M206` offsets are never used.

Since all moves in Marlin are handled in native machine space, and the mesh is also stored in terms of native machine space, offsets to the workspace have no effect on the position of the mesh relative to the physical bed.