MarlinFirmware/Marlin #17161: [BUG] Printer freezes on bigger prints.

Haxk20, 2020-03-14:
### Bug Description

We slice the model using Cura and transfer it to SDcard and start the print.
The prints starts out just fine but after some time (It ranges from 10 minutes to several HOURS) it just freezes. No movement on steppers and LCD screen just says the SDcard has been plugged in. But i know that nobody touched the SDcard (I have stood near the printer for the 10 minutes and saw it just stop).
This also weirdly enough happens with USB printing too. LCD says the same thing.
I had this happen on huge print (1.5 day) and it stopped after 16 hours.
I would be kinda OK if it stopped and bed and hotend went to room temp but they still stay heated up. This is a huge fire hazard.

### My Configurations
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4332947/Configuration.files.zip)
### Steps to Reproduce
1. Start a print
**Expected behavior:** Print starts and finishes just fine resoulting in printed model.

**Actual behavior:** Print starts and after some time freezes.

I also provide the gcode where the print stopped after 17 minutes:
[Gcode.zip](https://github.com/MarlinFirmware/Marlin/files/4332957/Gcode.zip)

Haxk20, 2020-03-16:
I have just tried deleting the entire SDcard. That yelded one vase mode print for me but failed on second one. @thinkyhead Opened a new issue as you requested.

Fantomiaso, 2020-03-16:
Have exactly same issue.
SKR1.3 + TMC2208 in uart mode. Stealthchop (LA in firmware is OFF). No display. Anycubic I3 mega hardware.
Prints both from onboard SD and via USB freezes at random point. Heaters, motors and and fan sremain ON. USB-Serial bridge stops responding, but after reconnecting USB I can send commands to Marlin again. I think that problem is in SKR hardware. When I have to test board outside of case any touch of board or wires coud halt it. Inside of metal case board is shielded, but some interference can occur. 
Tomorrow I will try better grounding of board and case. If it will not help, maybe I will add some capacitors on 5V 3V3 lines. In my opinion for reliable operation 12/24V should have 2000-3000uf, and 220uF at least on 5V and 3V3, but thera are only 100uF at 12/24V, 22uF at 3V3 and 33uF at 5V.

My config:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4340247/Config.zip)

thinkyhead, 2020-03-16:
When Marlin is itself paused or stopped it will send messages to host at regular intervals, so that is something to look for. Marlin will sit and wait if commands are no longer coming in from the SD card or from the host. There's currently no failover mode that deals with a hung up host or an SD card that stops responding. So that would be something new to work on.

martend, 2020-03-17:
I run skr 1.3 & 1.4 with matching tmc 2208's on 24v and don't run in to these problems. I use BTT stepsticks in uart mode, stealthchop etc. on. I also run one fan over the stepsticks and used the little potmeters to turn the factory current settings of 1.2 down to about 1.0 / 0.8. I observed while doing some prints if the sticks get to hot or there was to low or to much current on an axis (steppermotor) the print would fail. Other functions would continue like nothing happened, including the serial connection. Also a flimsy connection / cable could be the cause, our cheap hardware wants to be connected to good connectors and decent wires :-) I'm half through a 5.5 hour print now, later this evening I'm planning to start a 12h one, with the current Marlin firmware on the board. I'll report how it turned out.

Fantomiaso, 2020-03-17:
Added 470uF capacitors on 3V3 and 5V lines, 2200uF on 12/24V. Will try some prints for next 24 hours.

Fantomiaso, 2020-03-18:
Ok, things are getting more interesting. At first look, adding capacitors solved the problem, but after 4-5 hours print stops. Next attempts gives me random durations to failure event. 
Than I  updated firmware to the latest one and while migrating configs, I h changed
#define NO_TIMEOUTS 
with value 1000 print goes much further, but I noticed freezes with duration near one second. Finaly print also stopped. It was 2.5 hour print with near 8 freezes. Situation is the same with USB and SD prints. I'll try configs from other printers with working SKR 1.3 and marlin to find what can be wrong. Also I'll get one more SKR, but with 2209 to try.

martend, 2020-03-18:
@Fantomiaso could you share the gcode file you are printing? Would like to start a print and see what it does on my machine.

Fantomiaso, 2020-03-18:
[AI3M_Ai3MEGA-S_CaliperHolder_Right.zip](https://github.com/MarlinFirmware/Marlin/files/4350385/AI3M_Ai3MEGA-S_CaliperHolder_Right.zip)
> @Fantomiaso could you share the gcode file you are printing? Would like to start a print and see what it does on my machine.

makemerush, 2020-03-18:
I previously had this issue on SKR 1.4 Turbo with TMC2130 SPI mode for X/Y/Z and TMC2225 UART for E

Build 2.0.5.1 has fixed it for me! Not sure what changed that would have resolved it though. Recommend the new build if you haven't tried it yet.

Haxk20, 2020-03-18:
@makemerush The thing is this isnt new issue. Few people had this issue months ago. Then it out of nowhere stopped. And now its back. Its really crazy cause i have just done few prints and all of them went fine. And they werent quick prints. They were 5+ hours prints. So TBH i have no real clue as to where the issue is coming from.
Oh and did i forgot to say that the issue went away without reflashing the firmware or anything ?
Yeah just out of nowhere its gone it seems.
And worst is i never know when it will come back.

Fantomiaso, 2020-03-19:
Ok, I think I know what is the problem in my case. I started new print from SD via RepetierHost and detouched USB cable from printer (I don't have display at all, so I start prints from host). No freezes, no print stops. I tried different cables and hosts. In all cases, except one, problem turns back. That one case was when I used Odroid XU4 with battery power as host. My version of why that happens is bad USB implementation on hardware level. Maybe power interference, maybe signal. Don't have enough instrumentation to figure out with it. Will try USB print from Odroid next week, when I get stronger battery.

Haxk20, 2020-03-19:
Here i turned off StealthChop and the print is going fine for 2 hours so far. It is long print so will report if it goes to freeze or finishes.

Fantomiaso, 2020-03-19:
> 
> 
> Here i turned off StealthChop and the print is going fine for 2 hours so far. It is long print so will report if it goes to freeze or finishes.

There is a known problem with 2208 drivers. An extruder driver in StealthChop mode may freeze when using linear advance. Repraps wiki suggest using:
 #define MINIMUM_STEPPER_PULSE 2
(if notes work try 4)
When 2208 are user in "classic" mode (no uart), only the extruder driver freezes. I suspect that its freeze in the uart mode can lead to deadlocks when the response from this one driver never arrives. If the setting of the pulse duration does not help, it is enough to switch only the extruder driver to the spreadcycle and everything starts to work. But such a problem is not observed with the rest of the drivers and when the linear advanced is turned off.

Haxk20, 2020-03-19:
Im not running TMCs on Extruder due to this exact reason. I run it only on X and Y

Haxk20, 2020-03-19:
And the printer just froze.

martend, 2020-03-19:
Thanks for the file @Fantomiaso, will fire up that one later. 
I printed a nearly 6 hour file yesterday and it went well, but since my cold end was not playing nice I needed to fix that first before starting a long print. I'll start a 12h this evening.
@Haxk20 I noticed you and others use different models of stepperdrivers in your config. Running TMC's only on X and Y. I on the other hand have all axis populated with TMC 2208's and stealthchop on in uart mode. Could it be possible that different stepsticks config don't play nice together at the moment? If I would have any other drivers then TMC's to put in my board I could test along, but I don't have 'm.

Haxk20, 2020-03-19:
It is very much possible that there is something going on between them. Unfortunately i cannot test with different ones. I given  the original ones to friend.
But TBH it doesnt make sense. You are experiencing the same exact issue as us. Media insert on screen.
OK. Wait i just thought of something. This always started when i changed filament. TBH this at first sounded very crazy as how could filament cause this right ?
Well each filament is different in how much force they can handle before deformation occurs right ?
Well this happens in the extruder also as it need to grab the filament with teeth. And if the filament is grabbed stronger and the deformation is bigger or less then it can push more or less filament.
This in turn can make the motor work harder and require more current. One solution is to up the current but thats not the correct solution as the deformation is different the extruder is pushing different amount of filament. And well PETG and PLA are very different chemically vise. Thus the correct solution is to recalibrate the extruder steps per mm.

(I just came up with that and i may be horribly wrong. So i will tomorrow recalibrate and see what happens. And as safety precaution i will up the current to extruder.)

thinkyhead, 2020-03-19:
Are you able to force the printer to freeze by turning on heaters and doing fast printing moves? If you switch to "dry run" (or disable the heaters) does the freeze stop happening?

We've recently had an issue reported where an under-powered PSU experienced a voltage drop whenever the machine was printing at high speed with the heaters on (which you kinda need). If you are able to eliminate the problem of a drop in voltage at the input to the board, that will help narrow it down to… something else.

Haxk20, 2020-03-19:
> Are you able to force the printer to freeze by turning on heaters and doing fast printing moves? If you switch to "dry run" (or disable the heaters) does the freeze stop happening?
>
> We've recently [...]

It is well actually absolutely possible its voltage drop. Im technically running the bed at the limit of what PSU can deliver. The issue is i tried to order new PSU but well due to the world in its current state its just not possible to get it in here in the coming month.
But yes i will absolutely try to run it without heating up the printer and dry run it.

Haxk20, 2020-03-20:
Tested without heating and it finished fine @thinkyhead

Haxk20, 2020-03-20:
I will heat it up tomorrow and just not add any filament to see if that freezes too.

Haxk20, 2020-03-24:
@martend @makemerush @Fantomiaso Can you please tell me which slicer do you use ?
Cause i just tried one small print that crashed on me with CURA and now Prusa slice and in prusa slicer it printed. Trying a bigger print that crashed on me 100% of the times i tried it.

makemerush, 2020-03-24:
Cura 4.5 here

Fantomiaso, 2020-03-24:
In my case problem is not in software, but in combination of oversimplified schematics of SKR board (with no any protection except fuses) and lack of grounding. I'm living in Belarus and only about ten years ago making grounding in houses became mandatory. Mine is much mor older. I checked voltage between PC housing and 3D-printer and it was floating with peak near 80 volts. Even neon testing screwdriver glows when touching printer or pc, that tells us that it is either phase voltage or strong electromagnetic interference. That voltage latches MCU occasionaly. Same g-codes printed well from SD with USB detouched or via USB odroid single-board pc with battery-powered or fully isolated power source, but failed when printing when USB is connected to my PC.

Haxk20, 2020-03-24:
@Fantomiaso Can you tell me the slicer tho ?

Fantomiaso, 2020-03-24:
Slic3r and Cura 4.5

Haxk20, 2020-03-24:
Both of them produce freezes ? @Fantomiaso

martend, 2020-03-24:
@Haxk20 Most of the time Slic3r on a dedicated Linux box.
But since Marlin 2.02? I also started using Cura 4.5.0. Because since that update it seems my printer is on steroids with Slic3r, meaning it's ramping up the speeds like no other. When slicing with Cura it prints with the normal speeds like Slic3r did before.
I know it's a different thing, thought I would mention it anyway since the only thing changed is that Marlin got compiled with the newer version, together with the firmware for my BTT TFT V3.0 
[image]
Everything you see in red on the picture was printed the last 2 days, a 12h and 2 x 5h prints. Printed with Cura.

Fantomiaso, 2020-03-24:
@Haxk20 yeap. 
And as I wrote earlier, it's not they produce freezes (in my case).

Haxk20, 2020-03-24:
OK so we can exclude slicers i think. What voltage and amperage your PSUs have ?

martend, 2020-03-24:
220V what goes to a 360w 24V output 20A powerbrick. It's the stock one that comes with Anet ET4.

Haxk20, 2020-03-24:
I cant see how that could voltage drop. Also thats not 360W thats 480W.
Considering you have good cooling over the PSU i cant see how that could produce a voltage drop. @thinkyhead I cant see how 24V 20A PSU could voltage drop during heavy printing.

martend, 2020-03-24:
Got the specs from the site, can't look under the machine at the moment it's printing :-)

Haxk20, 2020-03-24:
Well seems like the OEM cant do simple math. :) Would not be the first or last time sadly.

Fantomiaso, 2020-03-24:
12V 350W PSU
Consumption: 200W buildplate + 40W hotend when cold. When heated resistans increases and working current become lower. 60W are for steppers. In worst case I have 50W in reserve.

Haxk20, 2020-03-24:
Yeah you both have much in reserve. I dont. I run 24V 8.5A (What came with printer). I cant even order a new PSU now due to state the world is in at the moment.

But considering both of you have a low of power still in reserve it seems unlikely that a voltage drop could appear. Well that is if the PSU is correctly cooled. Now that i think of it. Mine isnt. It could be that the printer heats up the PSU a lot over the hours and that could cause some voltage drop. I will print some feet for the printer and add a cooling fan to the PSU.

martend, 2020-03-24:
That might do the trick for now. Mail & packages still get delivered to me where I live, so I keep ordering things online :-) Maybe you have access to an old PC PSU those should be beefy enough also.

Haxk20, 2020-03-24:
Mail still arrives here but Ebay sellers are refusing to send it sometimes. Tried 2 times and both refused to send it due to the COVID-19

Fantomiaso, 2020-03-24:
> Well that is if the PSU is correctly cooled. Now that i think of it.

By the way wery possible. The transformer magnetic core changes its permeability sharply when overheated:
[image]

That means that at at temperatures over 90*C magnetic core of transformer requires higher frequency to operate at nominal power. But as switching frequency of PSU is fixed, you can really get powerdrop.

Haxk20, 2020-03-24:
@Fantomiaso I know electicity thats why i suggested the idea of PSU overheating.
Anyways. Can all of you tell me which SDcard do you use in SKR1.3 ? (Not for printing. But the one where firmware is uploaded)

martend, 2020-03-25:
I used the original, but now I am on a 32gb Lexar high speed one.

Fantomiaso, 2020-03-25:
I'm quite sure SKR stores firmware on chip. As for SDs... Any SD I see in that moment) I have plenty of them: from cheap no-name to branded SanDisk for nintendo.

amonpaike, 2020-04-07:
My experience is this: i have an anet A8 mostly original with a mosfet for heating the bed, and e3d v6 ..
I recently upgraded to marlin 2.0.5.2 after a couple of years, which I didn't upgrade the printer, more or less from version 1.1.6 ...
The printer works at its best, I had it printed heavily for a week 10 -15 hours a day .. it works very well, except that it freeze twice after 4-5  layers (Luckily).
When freeze, the display shows the temperatures at maximum, but the extruder and the bed cool down, and if I try to press the display it does nothing, the only way to make the printer refunction is to remove the power and turn it back on, I also discovered that the power loss restore doesn't even work (another bug).
I relaunch the same identical print and it is processed without other hitches.
Always print from SD card

martend, 2020-04-07:
@amonpaike pre-heating nozzle and bed might useful here, if you are not already doing that. Also check the temp settings in your slicer. Even since you upgraded, do a PID test for the nozzle and bed. Marlin"s temperature algorithm got stricter and better. 15,16 hours prints are normal for me also, without issues. See picture below.

martend, 2020-04-07:
21h plus print
[image]

amonpaike, 2020-04-07:
> @amonpaike pre-heating nozzle and bed might useful here, if you are not already doing that. Also check the temp settings in your slicer. Even since you upgraded, do a PID test for the nozzle and [...]

Thanks for the advice, I reported my experience because I am almost certain that this is a marlin bug (well who knows what)
I use slic3r PE, and throughout this week of prints I have only had this freeze two times and both times the exact same error, even with similar layer height (not the same amount of filament)
another detail that may be relevant is that throughout the week I used the exact same print settings in slic3r for all files

thinkyhead, 2020-04-08:
Since "freeze" is a very general thing that computers and software can do for a wide variety of reasons we ask that everyone experiencing a "freeze" post a new issue, filling out the full template, detailing your particular hardware, configuration, and so on, so that your specific "freeze" can be isolated and fixed. If it relates to another issue, that may be determined by further investigation.

amonpaike, 2020-04-10:
@thinkyhead  at the moment, since I wrote the previous comment, I have updated to the last marlin with a fine tuning with a PID test for the nozzle and bed as suggested by @martend , and I have been updating the parameters of my 3dprinter firmware configuration, for now I have not no longer had any freeze (and I hope it doesn't happen) I hope it is the solution, if time passes and I haven't reported any other freeze again, at least it can be relevant information..

martend, 2020-04-11:
@amonpaike awesome, glad it solved your issue. Your feedback is welcome and relevant since the other issues in this thread also got solved, this one can be closed if @thinkyhead agrees.

Haxk20, 2020-04-11:
I have started using octoPrint as when i printed with Cura via USB i saw no issues. And yeah octoPrint is the same. No issues. Also the bonuses are awesome you can get from OctoPrint.

Haxk20, 2020-04-11:
Forgot to say. Did 64 hours print. No freeze. This was using PETG on 240C and 85C.

Haxk20, 2020-04-11:
And considering PID tuning solved it for one user so far its almost surely PSU issue then.

martend, 2020-04-11:
@Haxk20 64 hours.. Nice!! Great Marlin and the board is also working for you and can only agree on Octoprint!

amonpaike, 2020-04-11:
@martend  I would wait a couple of weeks before closing this bug report, I have an intense week or two of other prints to do these days, and if no freeze will happen and I will not have reported it ... then at least "it is a possible cause and solution "  and a more solid confirmation.

xavsbl, 2020-04-14:
Got the exact same symptoms on my skr 1.4 turbo and Marlin 2.0.5.3. Didn't try fine PID tuning as I reported previous values of previous tuning but with a bugfix marlin version downloaded around february. So till now and since updating i observed that prints sometimes just stops after 4 to 5 hours printing, but sometimes the printer achieve its duty without problems.
Regarding PSU, it's a 24V/15A here but a somewhat "slim" version of what i'm used to see in CR10 clones (Sunlu S8 to be precise).
I'll report if the pid tuning does the trick.

xavsbl, 2020-04-15:
It doesn't. Had a 6h print stopping after 2 hours.

xavsbl, 2020-04-24:
Here reporting that my issues were not related to this thread. Actually the computer I use to control and monitor my printers has a usb controller issue. 
Sorry.

cooldudie2, 2020-05-30:
im having the same issue SKR mini E3 Board v1.2 intergrated tmc2209 drivers.printing from sd card Prints will freeze randomly at different points during the print. sometimes it doesn't freeze. can be hard to replicate as its random. Hot end powered and heated bed still powered fully on. the board is unresponsive to any input commands via pronterface or via screen. on;yl solution is to power off and backon again to get it to work.

things i have tried over the last month or so

1. tried cura and prusa slicer to rule out slicing settings/slicer issues
2. Different sd cards , manner of size ranges. always safety ejected from windows before use.
3. tried compiling BTT branch as well as latest marlin 2.0.5.3 same issues.
4. tried the 256k and 512k enviroments made no difference.
5. Linear advance and hybrid theshold modes are disabled to rule out those issues, still freezing.
6. had about 6 successful prints with no problems in a row between 4 -6 hours each problem has returned.
7. tried printing more slowly and dropping aceleration values,
8. tried disabling arc_per_segment, and adaptive step smoothing. stil persists.
9. tried PID tuning the hot end and bed error still persists.   
10. sent board back and returned and bought the board again from another supplier/distributor to rule out a bad board. issue has returned.
11. i havent tried setting it on fire and throwing it in the bin, thats probably going to be my next step.

martend, 2020-05-30:
How about your cooling? Besides the alu cooling blocks on the steppers, I
also putted on on the lpc1768 chip and I run a fan over the steppers/board.

cooldudie2, 2020-05-30:
yes i have the heatsinks on the stepper blocks and a fan running over them for cooling. I have also tried running the tmc debug feature in marlin whilst printing and I do not get any overheat warning triggers for the stepper drivers.

[25 more replies]

closed 2020-06-23 by boelle