MarlinFirmware/Marlin #17162: [Regression] M73 manual progress / remaining time ignored

pmjdebruijn, 2020-03-14:
### Bug Description

In very recent versions of Marlin, it seems M73 is ignored.

My suspicion would be #17082

### My Configurations

[skr_mini_e3_configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4332982/skr_mini_e3_configuration.zip)

### Steps to Reproduce

1. Generate a gcode file with M73 commands
2. Start print

**Expected behavior:** [What you expect to happen]
I'd expect to see to see the print time to finish start with a high number and slow decreasing to 0

**Actual behavior:** [What actually happens]
I start with 0, and see the print time slowly increasing

thinkyhead, 2020-03-15:
My suspicion would be b20d5a7cc627164877c95859baaa614f8afc7b99

thinkyhead, 2020-03-15:
Compare your configs to the latest `bugfix-2.0.x` configs and apply the latest changes from the patch above. Hopefully that is all we need to get it back in working shape.

commit 41ffe985 (2020-03-15): Fix LCD progress bar

Fixes #17162
closed 2020-03-16 by thinkyhead