MarlinFirmware/Marlin #17163: [BUG] Unable to set temperature on heaters when HOTENDS > EXTRUDER

marciot, 2020-03-14:
### Bug Description

I am working on a printer configuration where there is a single extruder with several heaters. I was able to set HOTENDS to 3 and EXTRUDERS to 1, however, when I try using "M104 S200 T1" I get an error "Invalid extruder 1". On the other hand, adjusting the heater through the EXTENSIBLE_UI allows me to set the temperature. So this appears to be an error in validation in M104, rather than an inability to set the temperature.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[CocoaPress_SingleExtruder.zip](https://github.com/MarlinFirmware/Marlin/files/4333235/CocoaPress_SingleExtruder.zip)

### Steps to Reproduce

1. Set HOTENDS to 3
2. Set EXTRUDERS to 1
3. Run "M104 S200 T1"

**Expected behavior:**

The second heater should begin to heat.

**Actual behavior:** [What actually happens]

"Invalid extruder 1"

marciot, 2020-03-14:
Possible solutions in `src/gcode/gcode.cpp`:

```
int8_t GcodeSuite::get_target_extruder_from_command() {
  if (parser.seenval('T')) {
    const int8_t e = parser.value_byte();
    #if HOTENDS > EXTRUDERS
       if (e < HOTENDS) return e;
    #else
       if (e < EXTRUDERS) return e;
    #endif
    SERIAL_ECHO_START();
    SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
    SERIAL_ECHOLNPAIR(" " MSG_INVALID_EXTRUDER " ", int(e));
    return -1;
  }
  return active_extruder;
}
```

Or, possibly extend `M104` to take an additional parameter `Hx` to manage hotends independently of extruders:

```
int8_t GcodeSuite::get_target_extruder_from_command() {
  if (parser.seenval('T')) {
    const int8_t e = parser.value_byte();
    if (e < EXTRUDERS) return e;
    SERIAL_ECHO_START();
    SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
    SERIAL_ECHOLNPAIR(" " MSG_INVALID_EXTRUDER " ", int(e));
    return -1;
  }
  if (parser.seenval('H')) {
    const int8_t e = parser.value_byte();
    if (e < HOTENDS) return e;
    SERIAL_ECHO_START();
    SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
    SERIAL_ECHOLNPAIR(" " MSG_INVALID_HOTEND " ", int(e));
    return -1;
  }
  return active_extruder;
}
```

boelle, 2020-06-20:
@marciot is the issue gone?

thinkyhead, 2020-08-04:
> I was able to set HOTENDS to 3 and EXTRUDERS to 1,

That is not a supported combination. You should instead set `EXTRUDERS` to the number of hotends you have, then invent a new type of extruder. I would recommend calling it `MULTINOZZLE` as a contrast to `SINGLENOZZLE`.

Then the next step is to add an entry to `Conditionals_LCD.h` which will be…

```cpp
  . . .
#elif ENABLED(MULTINOZZLE)
  #define E_STEPPERS 1
  #define HOTENDS EXTRUDERS
#endif
```
Then, if any other special behaviors are needed for `MULTINOZZLE`, well, it's Marlin, so you'll have to scatter those changes in the various places where they are needed, from the tool-change routine to the LCD display, and so on.

Basically, `EXTRUDERS` is a synonym for "tools" when applied to 3D printers, so that we can hide `HOTENDS` and `E_STEPPERS` and just use them internally according to the declared "style" of the extruder.

All of which is to say, we can and should still use `EXTRUDERS` in `get_target_extruder_from_command` because `EXTRUDERS` is used everywhere else, including tool-changing and LCD value editing, and so on. If you are using `HOTENDS` in any `get_target_extruder_from_command`-like code that more rightly refers to "the tool index" (where each "tool" in a `MULTINOZZLE` corresponds to a hotend), then it can and should be changed to `EXTRUDERS`.

The long-term plan with propagating extruder types and translating them into some general conditionals was and is to distill the common elements of extruders and develop some classes around them. Trying to make `EXTRUDERS` a proxy for the "tool index range" has been one aspect of that.

In contrast, extruders can get a little more complex, so then we have to add something like `MIXING_STEPPERS`, because the interface should present both a single E axis and a single nozzle, but we need custom menu items to select virtual tools and set the mix and so on.

I think your nozzle is less like `MIXING_EXTRUDER` and more like `MULTINOZZLE`, so there shouldn't be a lot of customizations needed, depending on whether the menus look reasonable with just the conditionals above.

If you don't like the appearance of the menus, but only some of them, then you may consider adding another conditional that describes how many of those specific things to display for `MULTINOZZLE`.

---
Sorry for neglecting this issue. It simply got buried in a vast pile at a time when the pile was much vaster.

marciot, 2020-08-05:
@thinkyhead: Just so you understand where I am coming from, this ticket came into play because I was adapting Marlin for the Cocoa Press chocolate printer. This printer has several reservoirs that must be kept to specific temps due to the finicky nature of chocolate. I guess what I would like to see in Marlin is the ability to enable "extra" heaters which aren't associated with a hot-end -- the hardware is already there, as is the control code, so it there simply needs to be an easier way to tell Marlin to allow those extra heaters to be enabled.

Anyhow, by disabling the validation code, I was able to get the behavior I wanted, allowing me to increase HOTENDS beyond EXTRUDERS, which gave M104 the ability to control those unused heaters. So I guess the functionality is there, it just maybe needs to be polished a bit.

thinkyhead, 2020-08-14:
Well, you know Marlin! Every addition is ad hoc, and there is just barely a general concept of a heater which is not tied to a specific well-defined use. Along with every other nook of the code, this is one area that could benefit from some refactoring, and I have been trying to do that in a piecemeal way as part of integrating other features.

To get custom heaters to the level a formal feature, the correct approach is not to try to repurpose "hotend" heaters or to expand on the extruder paradigm, but to work on, first, documenting the code as it currently exists so that it is very clear how it all works, and then to refactor the code to a point where it's generalized enough that adding new custom heaters is a trivial operation.

While the pins files and the Temperature code do right now strongly tie the 12V PWM pins to extruders and the bed, that could certainly be loosened up for custom applications. I would start by examining the way that the bed and/or chamber heaters and sensors code is partitioned from the other heaters, and look at the methods and behaviors that they share, and then just mock up a new "custom heater / sensor" combination based on those.

Ultimately, a general "custom heater / sensor" feature may not be of very much use. So typically when working on some custom addition like your extra chocolate melting reservoirs, I would integrate the new heaters just as if I was adding any other new and distinct hardware support. There would be an option called `RESERVOIR_HEATERS` and I'd define a `HEATER_RES_PIN` and a `TEMP_RES_PIN` in the machine's pins file, and then just I'd just add it into the mix alongside the chamber and bed, maybe give it a cute icon…

During that process, possibly I would end up doing a bit more work in the direction of generalizing heaters and making it easier to add the next custom heater. One thing is for sure. If I never dived into that code then I would never get up the momentum and knowledge to be able to refactor it….

closed 2020-09-22 by github-actions[bot]