MarlinFirmware/Marlin #17164: octoprint getting disconnected when print finishes

JWolfJr, 2020-03-15:
After compiling the latest firmware 2.0.5, Octoprint is getting disconnected when the print job finishes.  Had no issues with 2.0.4.
Current setup:
- Ender-3
- Bigtreetech SKR mini E3 V1.2
- Bigtreetech TFT35 V3.0
- Raspberrypi 3B+

Octoprint forum said something about a G4 command being issued when printing has finished.
I don't remember anything about G4 when comparing the new firmware to the last.
Can't find anything on the web, I imagine it's due to being a new issue with the latest version.

JWolfJr, 2020-03-15:
I noticed in my terminal log, M117 is the last command sent.  I just added it to my long running commands list for the connection not to close.

bill-orange, 2020-03-15:
And where might the "long running commands list for the connection not to close" be.  I would add it to mine if I could find it.

JWolfJr, 2020-03-15:
You can find it in settings > firmware & protocol > then under advanced options.

JWolfJr, 2020-03-15:
Today I just finished up a print.  After the print finished I get a M104 (set hotend temp), then M117 (for sending info to lcd screen), finally a M400 (Marlin docs: "This command causes all GCode processing to pause and wait in a loop until all moves in the planner are completed."). Then octoprint counts down and eventually times out and disconnects. M400 is also in my long running commands, placing M117 didn't help. @bill-orange any luck?

bill-orange, 2020-03-15:
It will be a few days before I can test.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle