MarlinFirmware/Marlin #17167: [FR] LCDs backlight control support ? (Screen brightness adjustment)

Lino77, 2020-03-15:
Prusa has a new display from LDO with which you can set the backlight. (from 3.9.0 Beta Screen brightness adjustment) this function is actually supper. you could add this function for the LCD2004, since not every display supports this function, introduce a new display LDO RepRapDiscount Smart Controller

ellensp, 2020-03-15:
This is already in marlin just define a LCD_BACKLIGHT_PIN

Lino77, 2020-03-15:
LCD BACKLIGHT PIN is set but does not get a menu entry like MKS mini e.g.

codiac2600, 2020-03-21:
After testing this won't work because all screens defined as capable of M250 are listed in CONDITIONALS_POST.h are graphical displays. 
Adding the REPRAP_DISCOUNT_SMART_CONTROLLER to the list wont compile.

There should be a new LCD added to the list for this to work. So a REPRAP_DISCOUNT_SMART_CONTROLLER with built in LCD controller chip. Until then this won't work.

tpruvot, 2021-10-21:
Was merged recently for the TFT but for now only in touchscreen native ones, maybe some class could be made for all screens with a backlight pin, and optionally with brightness control if its a PWM compatible pin, see also #16215