MarlinFirmware/Marlin #17168: [BUG] Double click needed to continue after finished SD print

Lord-Quake, 2020-03-15:
### Bug Description

Marlin 2.0.5

After a print from SD is complete a double click is needed to enter the Marlin menu.

**Expected behavior:** 

Only one click required to enter the Menu.

**Actual behavior:** 

One has to click twice to enter the menu in Marlin.

#### Additional Information

I believe this has to do with the issue https://github.com/MarlinFirmware/Marlin/issues/17106 we had where "Click to resume" was required. The message is not shown any more but the confirmation click is still present.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

Lord-Quake, 2020-06-23:
@boelle Thanks. Completely lost sight of this issue. It since has been resolved through a bug fix.