MarlinFirmware/Marlin #17169: [BUG] Cannot increase Z-Feedrate

Mosher23, 2020-03-15:
I have replaced my Z Lead screws from ones with 8mm lead to 1.5mm lead. I did adjust steps/mm and did some test cube prints. Print was perfect. However as my lead distance shortened my Z-speed dropped. So i tried increasing Z feedrate, but it always moves with the same slow speed. 

Am i on verge of some kind of physical/mechanical limits or is it a bug?

### Problem
Slow Z-speed independent from set feedrate

### My Configurations
Ender 3 dual Z 
SKR 1.4 Turbo TMC5160
Marlin 2.0.5

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/4333962/Configuration.Files.zip)

### Steps to Reproduce
replace Z lead screw to much smaller lead step

**Expected behavior:** [What you expect to happen]
increasing feedrate values for z axis will increase z movement speed
**Actual behavior:** [What actually happens]
increasing feedrate values have no effect on z speed

ellensp, 2020-03-15:
Z steps/mm 2133. is way to high. reduce you micro-stepping and this value to a more sane limit

Mosher23, 2020-03-15:
> Z steps/mm 2133. is way to high. reduce you micro-stepping and this value to a more sane limit

what is so insane about this?

ellensp, 2020-03-15:
eg feed rate 3mm/s thats 6399 steps every second, ie 6.4 kHz.   
eg feed rate 6mm/s that 12798 every second ie 12.8 kHz

Fairly sure marlin has a 8kHz stepping rate... 

so max you can do is about 3.7mm/s

Mosher23, 2020-03-15:
> eg feed rate 3mm/s thats 6399 steps every second, ie 6.4 kHz.
> eg feed rate 6mm/s that 12798 every second ie 12.8 kHz
> 
> Fairly sure marlin has a 8kHz stepping rate...
> 
> so max you can do is about 3.7mm/s

oh dang... now i get it! thank you so much!
Is there a way to increase stepping rate?

Mosher23, 2020-03-15:
@ellensp actually i've just checked default Configuration.h and it has 4000 steps/mm by default 
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }

ellensp, 2020-03-15:
and a max feed rate of 5, which needs lowered to 2 or 3 for most people as it doesn't work out of the box.

Mosher23, 2020-03-15:
@ellensp i had it on defaults before, when i used lead screw with 8mm lead and it was running much much faster

ellensp, 2020-03-15:
8mm lead is 400 steps/mm on most machines. not 4000

Mosher23, 2020-03-15:
@ellensp ah ok. thank you again

psavva, 2020-03-17:
Please close this ticket if your issue is resolved.

closed 2020-03-17 by Mosher23