MarlinFirmware/Marlin #17172: [BUG] Marlin-2.0.5 binary a lot bigger than 2.0.4.4 with same config

Marlor, 2020-03-15:
### Bug Description

The Marlin-2.0.5 binary is a hefty chunk bigger than in 2.0.4.4 with the same config
Is this a bug or intended?

The only differences in the configs are things like,
the renaming from HAS_TRINAMIC to HAS_TRINAMIC_CONFIG etc.

### My Configurations

[Marlin-2.0.4.4.zip](https://github.com/MarlinFirmware/Marlin/files/4334695/Marlin-2.0.4.4.zip)
firmware.bin 261 KB
```
RAM:   [========  ]  76.9% (used 25184 bytes from 32736 bytes)
Flash: [======    ]  56.1% (used 266380 bytes from 475136 bytes)
```

[Marlin-2.0.5.zip](https://github.com/MarlinFirmware/Marlin/files/4334697/Marlin-2.0.5.zip)
firmware.bin 326 KB
```
RAM:   [=======   ]  69.4% (used 22728 bytes from 32736 bytes)
Flash: [=======   ]  70.2% (used 333452 bytes from 475136 bytes)

```

thinkyhead, 2020-03-15:
Which feature is the one that —when turned off— shows no significant size difference?

ellensp, 2020-03-15:
Interesting. the firmware.elf files are about the same. but the bin file generated from the .elf is way larger

Marlor, 2020-03-15:
The tests i could do today:

**default Marlin Config:**
Marlin-2.0.4.4 
firmware.hex **90 KB**

Marlin-2.0.5
firmware.hex **90 KB**

**default CR10s Config:**
Marlin-2.0.4.4 
firmware.hex **434 KB**

Marlin-2.0.5
firmware.hex **435 KB**

**default Marlin Config only change BOARD_BTT_SKR_V1_3:**
Marlin-2.0.4.4 
firmware.bin **85 KB**

Marlin-2.0.5
firmware.bin **85 KB**

**default CR10s Config only change BOARD_BTT_SKR_V1_3:**
Marlin-2.0.4.4 
firmware.bin **169 KB**

Marlin-2.0.5
firmware.bin **235 KB**

So it has to be a feature which is off in default, on in default CR10s
And the bloat does only happen with a LPC1768 Board (maby more) but not on Atmega

thinkyhead, 2020-03-15:
I've been testing build sizes, and here's what I can tell you…

With Marlin 2.0.5 if you want to use `FLASH_EEPROM_EMULATION` you have to specify it, otherwise it now falls back to `SDCARD_EEPROM_EMULATION` for LPC1768.

Marlor, 2020-03-15:
Sounds like an easy fix, add

```
// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
```
to `pins_BTT_SKR_V1_3.h`

But maybe more boards have slipped through the cracks

thinkyhead, 2020-03-15:
Sure thing. I didn't want to presume which boards should get which form of emulation by default. It seems most sensible to fall back to SD because this is the least volatile.

Previously with 2.0.4 you couldn't choose SD-based EEPROM emulation at all because the `persistent_store_api.h` file always defines `FLASH_EEPROM_EMULATION`. So it is now possible to choose, but yep, the fallback is SD.

closed 2020-03-15 by thinkyhead

thisiskeithb, 2020-03-16:
@thinkyhead I saw that an eeprom.dat file was generated on my SD card as if I switched to using `SDCARD_EEPROM_EMULATION` with the latest bugfix-2.0.x on an SKR 1.3 build I'm working on, so I looked into it and found this issue. I can open another one if this won't be reopened.

> ```
> // Use one of these or SDCard-based Emulation will be used
> //#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
> #define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
> ```

This doesn't compile on the SKR 1.3 when added to `pins_BTT_SKR_V1_3.h` with `FLASH_EEPROM_EMULATION` enabled:  https://pastebin.com/YdVSL9vw

`SRAM_EEPROM_EMULATION` compiles fine, but also generates an eeprom.dat file on the SD card and won't work without it:
```
Send: M500
Recv: echo:No EEPROM.
```

thinkyhead, 2020-03-16:
Patched with #17184