MarlinFirmware/Marlin #17178: [BUG] (short description)

SPRYGUYUK, 2020-03-16:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

SPRYGUYUK, 2020-03-16:
Hi all, i tried today to get bugfix 2.0 installed on my SKR v1.3, i am using VS Code to compile, i have LPC1768 set as the enviroment in platformio.ini and i copied/pasted from the boards.h  the #define BOARD_BTT_SKR_V1_3 and placed it in Configuration.h. but when i go to build/upload i get the following error and it fails.

Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:666:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
     #error "Unknown MOTHERBOARD value set in Configuration.h"

thisiskeithb, 2020-03-16:
Can you follow the issue template and upload your configs? I just compiled my SKR 1.3 build with `2.0.5` and `bugfix-2.0.x`, so it's probably a typo.

```cpp
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V1_3
#endif
```

ellensp, 2020-03-16:
Probably deleted to much and thier Configuration.h no reads
```
#ifndef MOTHERBOARD
  #define BOARD_BTT_SKR_V1_3
#endif
```
instead of #define MOTHERBOARD BOARD_BTT_SKR_V1_3

thisiskeithb, 2020-03-16:
> Probably deleted to much and their Configuration.h now reads

Yep. That's why I pasted the correct block.

SPRYGUYUK, 2020-03-16:
Morning, you was absolutly right, the boards.h file is missing mother off all its boards definitions and i had not noticed, replaced it and it is now working correctly in that reguard, but now i am getting more compile errors because of missing opperands.

In file included from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:0:
Marlin\src\HAL\LPC1768\../../core/macros.h:167:37: error: missing binary operator before token "("
 #define _DO_7(W,C,A,V...)  (_##W##_1(A) C _DO_6(W,C,V))

I have got 11 of these mostly from the same macros.h file, i have not edited this file, but its stopping it from compiling

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4337224/Marlin.zip)
i have included the config files and the macros.h

ellensp, 2020-03-16:
Your Configuration.h is still wrong.

you have "#define MOTHERBOARD_BTT_SKR_V1_3"
you **NEED** "#define MOTHERBOARD BOARD_BTT_SKR_V1_3"

It's **THREE** keywords not two.

**#define  + {space character} + MOTHERBOARD  + {space character} + BOARD_BTT_SKR_V1_3**

SPRYGUYUK, 2020-03-16:
thanks for that, i had spotted it and corrected it, its all working now, just have z probe and end stops left to sort out

closed 2020-03-16 by thinkyhead