MarlinFirmware/Marlin #17179: [BUG] Flash-based EEPROM no longer works on SKR 1.3

thisiskeithb, 2020-03-16:
### Bug Description

I saw that an `eeprom.dat` file was generated on my SD card as if I switched to using `SDCARD_EEPROM_EMULATION` with the latest bugfix-2.0.x (a662079) on an SKR 1.3 build.

Reverting to `2.0.4.4` allows flash-based EEPROM to work correctly again (and no `eeprom.dat` file is generated).

The following was suggested in https://github.com/MarlinFirmware/Marlin/issues/17172:
```
// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
```

This doesn't compile on the SKR 1.3 when added to `pins_BTT_SKR_V1_3.h` with `FLASH_EEPROM_EMULATION` enabled in `bugfix-2.0.x` (a662079): [Full error log](https://pastebin.com/YdVSL9vw).

`SRAM_EEPROM_EMULATION` compiles fine, but also generates an `eeprom.dat` file on the SD card and won't work without it:
```
Send: M500
Recv: echo:No EEPROM.
```

### My Configurations

`2.0.4.4` configs with flash-based EEPROM working correctly:
[myskr13-Marlin-2044.zip](https://github.com/MarlinFirmware/Marlin/files/4336070/myskr13-Marlin-2044.zip)

`bugfix-2.0.x` (a662079) configs that use SD card/`eeprom.dat` no matter what:
[myskr13-Marlin-bugfix-a662079.zip](https://github.com/MarlinFirmware/Marlin/files/4336071/myskr13-Marlin-bugfix-a662079.zip)

### Steps to Reproduce

1. See above.

**Expected behavior:** Flash-based EEPROM to work like it did in 2.0.4.4

**Actual behavior:** SD card-based `eeprom.dat` is generated no matter what.

Marlor, 2020-03-16:
Can confirm `#define FLASH_EEPROM_EMULATION` in `pins_BTT_SKR_V1_3.h` gets a compile error.
But setting it in `Configuration_adv.h` compiles fine and is working as intended, no `eeprom.dat` etc

thinkyhead, 2020-03-16:
What is the best choice for EEPROM fallback? There seems to be some question whether it's better to use the onboard SD when available, versus something like Flash-based. What should the order of preference be?

Previously, you couldn't select anything other than `FLASH_EEPROM_EMULATION` for `LPC1768` and yet there was recent work on the SD stuff.

thinkyhead, 2020-03-16:
See #17184

closed 2020-03-16 by thinkyhead