MarlinFirmware/Marlin #17180: [FR] (Auto shut down on 3D print end)

infatum1, 2020-03-16:
### Description

Requiring automatic shut down after a long print ends. Leaving home from work on Friday with 10-14 hour prints in the process ending at night. Coming back Monday. The printer basically stays in idle work mode for almost 2 days wasting electricity, producing fan noise and potentially being a fire hazard.

### Feature Workflow

1. Print ends.
2. Nozzle auto homes.
3. Motors deactivate.
4. Fans work until thermistors drop the temperature to about 30 C.  Bed drops to 30 C.
5. PSU auto shuts down.

#### Additional Information
I tried playing with firmware and adding the following to configuration.adv.h and got an error in Arduino IDE (expected constructor, destructor, or type conversion before '(' token). Checked all the brackets and tried different IF statements in C/C++. To no avail. 

```
//custom code to shut down 3D printer when the print is finished. Takes into account REMAINING_TIME and TEMP_SENSOR readings below 30 C. Then it turns off the 3D printer.

#if (REMAINING_TIME <= 0 && TEMP_SENSOR_0 <= 30 && TEMP_SENSOR_1 <=30 && TEMP_SENSOR_2 <=30 && TEMP_SENSOR_3 <=30 && TEMP_SENSOR_4 <=30 && TEMP_SENSOR_5 <=30 && TEMP_SENSOR_6 <=30 && TEMP_SENSOR_7 <=30 && TEMP_SENSOR_BED <=30 && TEMP_SENSOR_PROBE <=30) 
 WRITE(PS_ON_PIN, LOW); 
```

I would also be glad if this feature was added to the LCD main menu as an option.

Marlor, 2020-03-16:
Already possible:
1+2+3. Nozzle park and Motor off is in end gcode
4. see `#define USE_CONTROLLER_FAN` and `#define E0_AUTO_FAN_PIN`
5. see `#define PSU_CONTROL`

The limiting factor is your hardware not the software

infatum1, 2020-03-16:
Can you please walk me through the point 5?

We take Configuration.h file [https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration.h#L324](url)
We find the following: 
```
//#define PSU_CONTROL
//#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_HIGH false     // Set 'false' for ATX, 'true' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 100   // (ms) Delay for the PSU to warm up to full power

  //#define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif
#endif`
```

Which line needs to be touched?

ogsv, 2020-03-16:
#define AUTO_POWER_E_TEMP 50
Replace 50 with whatever temp you want to cool down to before switching off the PSU.

Sebazzz, 2021-04-23:
@ogsv I don't think that works in scenarios where the relay is also powering the power supply that runs Marlin, and needs to have a HIGH signal at all times except for the case when you want to shut down. There is SUICIDE_PIN of course, but that doesn't come with all the extras.

slowbro, 2021-07-07:
This is already implemented, as others have said. Enabling `AUTO_POWER_CONTROL`, and setting `POWER_TIMEOUT` to whatever value you want. 

For `AUTO_POWER_E_TEMP`, a better description for this is 'keep power supply on if any extruder is above this temperature'. So, you'd set this to 30.

closed 2021-07-07 by thisiskeithb