MarlinFirmware/Marlin #17182: Anycubic Kossel Delta Plus, still can't make it work properly

pablopeu, 2020-03-16:
Been trying latest releases with no luck, I keep going back to dec/2019 Bugfix2.0
The procedure is as follows:
Flash firmware (via octoprint)
From here everything is done from the LCD menu and encoder
Init eeprom
install probe
run auto calibration
remove probe
store settings
verify with paper if nozzle grabs paper
if not, take note of value and correct Z in delta settings accordingly
store settings
bed leveling
enable bed leveling
store settings
auto home
run a test print

The end result is an improperly calibrated bed.

If the calibration procedure is wrong, please let me know and I do it again.

A suggestion for the example files, speed of 6000 should be 2000 for the printer to move smoothly, acceleration should be 2000 instead of 3000 for the same reason. I have the Jerk value at 6 instead of the default 5.

Besides that I ran the config files almost default, here are them attached.

[Marlin 2.05.zip](https://github.com/MarlinFirmware/Marlin/files/4339822/Marlin.2.05.zip)

pablopeu, 2020-03-17:
Installed 2.051 did the following:

Flash firmware
init eeprom
deploy probe
auto calibration
store settings
LCD panel RESET
auto home
soft endstops OFF
test nozle Z0 with paper, take note of value if not zero (I had to adjust)
substract this value from Delta Height.
store settings
LCD panel RESET
auto home
deploy probe
bed leveling
store settings
LCD panel RESET
bed leveling ON
store settings
auto home

When I measure with a paper moving towards the bed I get a 0.75 height difference when I should get zero.

I'm open to suggestions or calibration procedures.

pablopeu, 2020-03-17:
changed to Unified Bed Leveling instead of the default bilinear, did the automatic mesh calibration, not perfect, but I was finally able to print a test circle across the whole bed, Now printing a benchy at the center of the table.

pablopeu, 2020-03-18:
keeps failing the prints... :(

wyattearp, 2020-04-05:
FYI - I just did this today and I'm having success on Marlin 2.0.5.3. Attached are my configs for reference (Linear Plus). I'm still working through it, but I also used the calibration steps here: http://www.lpomykal.cz/3d-printers/kossel/kossel-bed-calibration-marlin-fw/ (which is also where I ported the configuration from)

[AnyCubicKLP.zip](https://github.com/MarlinFirmware/Marlin/files/4434898/AnyCubicKLP.zip)

sjasonsmith, 2020-04-05:
> The other "Calibrate X/Y/Z" items are not having a good time though. Looks like hard coded values in the `menu_delta_calibrate.cpp`

@wyattearp, can you elaborate on what you mean by this?

pablopeu, 2020-04-05:
The minute I leave the mandatory countrywide quarantine (Argentina) I will check that config. Thanks

wyattearp, 2020-04-05:
@sjasonsmith sure - I accidentally clicked the "Calibrate X" menu item in a fried 2.0.5.3 build. When I did this, the printer head moved to Y0, Z0, and X-smash into the tower. Looking in the `menu_delta_calibrate.cpp` I see the following that would clearly be well outside of my print bed ranges:

```
  void _goto_tower_x() { _goto_tower_a(210); }
  void _goto_tower_y() { _goto_tower_a(330); }
  void _goto_tower_z() { _goto_tower_a( 90); }
  void _goto_center()  { xy_pos_t ctr{0}; _man_probe_pt(ctr); }
```

I've not read enough of the code to understand what it's doing yet, but those look like the angles I would use to calibrate, but also I keep seeing stuff in the code that is "everything in deltas has to be the same" and the math makes me wonder if that's getting messed up via some other thing.

thinkyhead, 2020-04-05:
90 degrees plus 120 degrees is 210 degrees. Plus 120 degrees is 330 degrees. Towers are located at these positions on the circle.

wyattearp, 2020-04-05:
Yea - I see that now. I'm still debugging my gcode weirdness (which is I think off-topic for this issue).

boelle, 2020-06-20:
@pablopeu is issue gone?

pablopeu, 2020-06-21:
Completely gone after upgrading to 2.053 bugfix from June/9 
Using the provided config with small touches here and there my Anycubic Kossel Delta Plus printer is working great again since 1.1.9
Here is my config, if it works for you, please let us know. cheers

My config: 
[Marlin-BugFix_June_9.zip](https://github.com/MarlinFirmware/Marlin/files/4809761/Marlin-BugFix_June_9.zip)

boelle, 2020-06-21:
i dont have a delta, but if the issue is gone for you maybe we should close this one?

pablopeu, 2020-06-21:
You're right Sir. Closed it is.

closed 2020-06-21 by pablopeu