MarlinFirmware/Marlin #17189: SKR 1.3 hangs when a heated bed is added to config

touwen75, 2020-03-16:
Hello, newbee here...

I have designed and build a Corexy printer with two extruders and a heated bed. 
The machine is controlled by a BTT SKR 1.3 with a TFT35 v3 display and 5 TMC2130 SPI motor drivers. After fiddeling/youtubing/googeling for two weeks now, I have got the display running, I can move all axis and both extruder steppermotors, and the Sensorlesshoming works for X,Y and Z. But I can not get the thermistor for the heated bed running...
 
If I defined both extruders and the bed with a dummy sensor, like so:
#define TEMP_SENSOR_0 999
#define TEMP_SENSOR_1 999
#define TEMP_SENSOR_BED 999

then everyting works; the display starts up, connects to the mainboard, and I can read the dummy temperatures. If I rais the temp a led starts blinking, so the heaters are on, and all the steppers can be moved.

But as soon as I config the Bed with a real sensor like so:
#define TEMP_SENSOR_BED 1
And connect one a pt100 to the right pins, the SKR 1.3 refuses to boot, a the display tells me there is "no printer attached"

I have tried turning all kinds of settings on and off, like: pidtemp, pidtempbed, all thermalprotections, watchdog, removing all sensor from the board.  
I have tried the BTT marlin branch for the skr 1.3, and the marlin 2.0.x firmware with the config 020004 and config 020005 files.
And now I am out of ideas.....

### Bug Description

SKR 1.3 hangs on boot if config ed with a bed temp sensor.

### Steps to Reproduce

1. configure; #define TEMP_SENSOR_BED 1
2. build the firmware.bin for a LPC1768
3. upload to SD card
4. place sd card in sd reader of the SKR 1.3 board
5. turn on 12v power
6. wait a few second te alllow the tft35 display to connect the SKR 1.3

**Expected behavior:** 
I expect the display to connect to the board.

**Actual behavior:** 
Instead the tft35 display keeps showing the error "no printer attached"

#### Additional Information

I have added the config files that does work, untill  you add : #define TEMP_SENSOR_BED 1
[Configuration that does work.zip](https://github.com/MarlinFirmware/Marlin/files/4340710/Configuration.that.does.work.zip)

AnHardt, 2020-03-16:
A 'PT100' is not a '1 : 100k thermistor - best choice for EPCOS 100k (4.7k pullup)'.

A PT100 has at room temperature about a resistance of 100 Ohm. A 100K thermistor about 100.000 Ohm.
So Marlin sees a nearly shorted thermistor if configured to '1' - and stops she system.

Better connect a thermistor and configure the right one or configure '147 : Pt100 with 4k7 pullup' even if that will not result in a really usable system.

touwen75, 2020-03-17:
Thank you for your comment!
I now know what went wrong:
1- I am calling my sensor a pt100 but it is infact a 100k thermistor: https://www.123-3d.nl/search/?search=DTH00002, so I have the right one with the right settings. but...
2- before I hooked up this one, I accidentally hooked up a k type thermocouple, it has a very low resistance (about 10 ohm). This fried the SKR 1.3,  The thermistor pins for the read 3.3 volt, the thermistor pins for the bed read 0.01 volt. 

So I guess I need a new board...

ellensp, 2020-03-21:
Since this isn't a firmware bug, can you close this issue?

closed 2020-03-21 by touwen75