MarlinFirmware/Marlin #17190: [BUG] Possible bug with filament runout detection

kingofmonkeys, 2020-03-17:
Description

I have been trying to get filament runout working for the past few days on what was at the time the latest bugfix branch.  I cannot get it to trigger during a print, using M119 it shows triggered and open correctly and the sensor is on during prints but no matter which way the switch is the printer does not detect a runout condition.  However if I enable #define FILAMENT_RUNOUT_DISTANCE_MM 1 it runs fine. 

My Configurations - attached config files. 

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4341299/Configurations.zip)

Steps to Reproduce

1. Turn on filament runout protection (define FILAMENT_RUNOUT_SENSOR)
2. Leave #define FILAMENT_RUNOUT_DISTANCE_MM commented out
3. Create out of filament condition while printing

**Expected behavior:** Expect filament runout condition to be detected and m600 run. 

**Actual behavior:** Nothing

Additional Information

Just to clear up a few common things that I know can affect this.  Yes I have Octoprint but I was printing via an SD card for these tests.  I know about FIL_RUNOUT_PULLDOWN and PULLUP, i have tried with and without them set (I don't believe i need them since its a SPDT switch but tried them just in case).  I can go from a non-working state to a working state by simple uncommenting #define FILAMENT_RUNOUT_DISTANCE_MM

I have attempted to look at the code in runout.h and debug some myself but I am slightly over my head, I believe the issue is with the debounce code since that is the only part switched when #define FILAMENT_RUNOUT_DISTANCE_MM is not defined but i can't tell if reset() or filament_present are being called erroneously or if it is something completely different.

kingofmonkeys, 2020-03-17:
So I added some debug output in an attempt to troubleshoot the issue and as soon as I added some debugging it started working correctly.  I remove the debugging and it once again doesn't work.

BastR, 2020-03-25:
Same problem for me, I can see OPEN or TRIGGERED with M119 but the printer does nothing..

BastR, 2020-03-25:
Ok, made some tests:

SKR PRO, Prusa Filament Sensor, FIL_RUNOUT_PIN on pin PA2

With #define PINS_DEBUGGING activated, I have:
```
[...]
Send: M43
Recv: PIN: PA2  (A2)   M42 P37     FIL_RUNOUT_PIN               Input  = 1
```

With M43 PA2 W, and triggering myself the sensor, I have:
```
[...]
Send: M43 PA2 W
Recv: Watching pins
Recv: PIN: PA1  (A1)   M42 P36          SERVO0_PIN
Recv: PIN: PA2  (A2)   M42 P37          FIL_RUNOUT_PIN
Recv: PIN: PA2  (A2)   M42 P37          FIL_RUNOUT_PIN
```

I am not able to debug any more, but marlin sees the PIN state change correctly but does not call M600.

@kingofmonkeys I think you can change the title and remove the "possible"

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle