MarlinFirmware/Marlin #17192: [BUG] POWER LOSS on Sbase delta

karabas2011, 2020-03-17:
### Bug Description

Turn on the printer after middle print turning off.

1 No any boot messages about Resuming.

2 I use User menu command and start it (see below)
Printer go homed than wait for temps and goes XY position loosing steps.
Ok I comment out G28 from G1000.cpp and increese z feedrate from 200 to 4000.
Retry.
Now home, z down  -70, then resumes XY and goes  to resumed Z. 
OK!
then stops and homes and starts the gcode file from the begining!!!
So M24 does not get position from PLR file.

3 I forced to eject  microsd from the board because I see it in menu insteed of LCD SDCARD !!!
And naturally no PLR on it.

### My Configurations
Delta printer
Marlin 2.0x from 05.03.2020
Sbase
No POWER LOSS pin !
User menu: added command "G28\n G91 Z-70 F4000\n G90 \n M1000 S"

### Steps to Reproduce

thinkyhead, 2020-03-18:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle