MarlinFirmware/Marlin #17193: [BUG] (short description)

motrj3, 2020-03-17:
I installed my bltouch a couple weekends ago and have been trying to iron out errors and bugs ever since. I was hoping to get some help on this one since I can't seem to find anything on my own. The issue I'm currently having is when I auto home, the x and y find the stop switches then immediately max out the opposite direction to home z instead of stopping in the middle of the bed. I was able to stop this by making the bed size smaller in firmware but have no idea how to truly fix the situation. I appreciate any help and have attached my config and config adv files.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4343148/Configuration.zip)

qwewer0, 2020-03-17:
[Guide]( https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/ ) for the board.

ellensp, 2020-03-17:
If bed size is correct and it moves to far you need to increase your micro stepping settings. 
Your config has driver type commented out so that defaults to A4988, this is not correct. You need to set driver types to TMC2209 by removing the preceding "//" 
After that you can adjust the micro stepping in Configuration_adv.h  eg     #define X_MICROSTEPS     16

thisiskeithb, 2020-03-17:
[Marlin's Configuration repo](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2) has configs for the SKR Mini E3 1.2 (Ender-3) that has everything setup for you.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle