MarlinFirmware/Marlin #17196: [BUG] (Dual Z endstop skr 1.4 stop all z axis)

lucableve, 2020-03-17:
Hi,
I updated firmware on my skr 1.4, when triggered z1_min or z2_min  all axis z stopped simultaneally.

My config is :
`
Configuration.h
#define Z2_DRIVER_TYPE TMC2209
`

`
Configuration_adv.h

 #define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many
#if NUM_Z_STEPPER_DRIVERS > 1
  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _ZMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
  #endif
#endif

`

`
pins_BTT_SKR_V1_4.h
#define X_DIAG_PIN         P1_29   // X-STOP
#define Y_DIAG_PIN         P1_28   // Y-STOP
#define Z_DIAG_PIN         P1_27   // Z-STOP
#define Z_MAX_PIN          P1_25  // PWRDET
#define E0_DIAG_PIN        P1_26   // E0DET
#define E1_DIAG_PIN        -1   // E1DET
`

I have checked endstop trigger by M119 and all works but no have idea how to fix this problem

ellensp, 2020-03-17:
What type of endstops?
Are you aware you need to modify your stepper drivers and remove the DIAG pin if you use switch based endstops?

lucableve, 2020-03-17:
I use switch endstop and have cut diag pin on all step driver

lucableve, 2020-03-17:
Just for information, initially even if the endpoints were triggered, the engines didn't stop. I had to enable ENDSTOPS_ALWAYS_ON_DEFAULT

lucableve, 2020-03-17:
This is my config.
My printer is anycubic i3 mega with Double z and double endstop switch.

On skr 1.3 and marlin version 2.0.0 work all
[Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4345849/Marlin.Configs.zip)

closed 2020-03-29 by lucableve