MarlinFirmware/Marlin #17197: [BUG] Fan soft PWM stops extrusion [Hemera]

Mosher23, 2020-03-17:
### Bug Description

My part cooling fan is set to start cooling from layer 3. As soon as fan start spinning my extruder [e3d hemera] stops extruding, even the gcode is being send.
I disabled fan_soft_pwm and now all is working, but i can’t control fan speed.

### My Configurations

Marlin 2.0.5
Skr 1.4 Turbo
E3D Hemera

### Steps to Reproduce
1. set Fan_Soft_pwm in Marlin
2. Use Hemera as extruder (didn‘t verify with other extruders)
3. Print
4. when the part cooling starts, extruder stops

psavva, 2020-03-17:
Please attach both your config files

Mosher23, 2020-03-17:
Sorry, here they are
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/4345367/Config.files.zip)

Mosher23, 2020-03-25:
No help? 
I disabled Fan soft PWM and it still stops extrusion.

swahananda, 2020-03-26:
Initially I was going to make a separate issue since your initial comment said the extruder would work with fan_soft_pwm disabled, and my Skr Pro v1.1 has been disabling E0 upon pwm fan activation regardless of the soft_pwm configs, but your comment from yesterday is in line with my experience so I will hop on this thread and offer to keep the issue consolidated unless requested.

The last functioning revision was 1.0.1 but I was not actively tinkering with the printer for 1.0.2-1.0.4 releases. 

I’m currently away from the computer but I will post zipped configs when I can. Though I have tried both BTT TMC5160 v1.2’s and Watterott TMC5160 v1.1’s that have been verified to be functioning.

I can update with verbatim output of M122 output as well, but for now I can say that all drivers are shown as enabled prior to PWM fan being enabled and then E0 showing as “false” when the fan kicks in.

boelle, 2020-06-20:
@Mosher23 is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle