MarlinFirmware/Marlin #17199: [BUG] Switching Nozzle X Y Offset Ignored

Cultivate3D-Dan, 2020-03-18:
### My Configurations

[config-no-offset.zip](https://github.com/MarlinFirmware/Marlin/files/4347169/config-no-offset.zip)

### Steps to Reproduce

Latest Bug Fix 2.0 Branch :
1) Home Printer G28
2) G1 X50 Y50
3) T1 > Z Lift occurs, Servo Switches does not move X and Y offset
4) T0 > Z Lift occurs, Servo Switches does not move X and Y offset
**Expected behavior:** X and Y Offsets are observed on tool change as per HOTEND_OFFSET_X and HOTEND_OFFSET_Y

**Actual behavior:** No X and Y Offset occurs on tool change

#### Additional Information

Is a simple setup with switching nozzle (****not extruder***) controlled by a single servo.

c3D-Dan, 2020-03-18:
Ignore please.  M502 M500 seemed to get things working.  Onto next bug.

c3D-Dan, 2020-03-18:
Closed this too soon.  Managed to find a work around, but its weird.  So From printer power up, T0 and T1 yeilds no offset movement.  Run M502 M500 (maybe not necessary) then M218 T1 X50 Y50 Z20.  Now try to move X or Y gantry, printer error 999.  Move gantry immediately and moves fine.  Now run T1 and everything working perfectly. T0 same.    If I turn off or restart the printer, I have to repeat this process.  See log below :

16:23:44.884 : echo:DEBUG:INFO,ERRORS
16:23:51.522 : N16 M502*19
16:23:52.293 : echo:Hardcoded Default Settings Loaded
16:23:53.345 : N17 M500*16
16:23:53.371 : echo:Settings Stored (707 bytes; crc 35197)
16:24:02.079 : N18 M218 T1 X50 Y50 Z20*45
16:24:04.506 : N19 G1 X-10 F4800*46
16:24:04.537 : Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
16:24:07.206 : N20 M999*24
16:24:07.206 : N1 M110*34
16:24:07.206 : Resend: 2
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:08.221 : N1 M110*34
16:24:09.375 : N2 G28*17
16:24:09.376 : echo:Active Extruder: 0
16:24:25.727 : echo:Active Extruder: 0
16:24:25.727 : X:50.00 Y:50.00 Z:0.00 E:0.00 Count X:64000 Y:64000 Z:0
16:24:32.736 : N3 T0*57
16:24:32.736 : echo:Active Extruder: 0
16:24:34.519 : N4 T1*63
16:24:36.601 : echo:Active Extruder: 1
16:24:39.079 : N5 T0*63
16:24:41.963 : echo:Active Extruder: 0

closed 2020-03-19 by c3D-Dan

c3D-Dan, 2020-03-19:
Running M999 as soon as printer turns on fixes this behavior and everything works as expected thereafter.  Now trying to find the cause of the Error.  Will document what i find for posterity.