MarlinFirmware/Marlin #17202: [BUG] Enable for all drivers goes low mid print

TouffeTouffe, 2020-03-18:
I'm using the latest bugfix as of March 17th.
X, Y and Z are TMC 5160s v1.4 and E0 is an A4988. This is a "dondolo" printer (dual hotend, one extruder motor and driver). I'm using a genuine Arduino Mega and a ramps 1.4 EEB.

At random, the printer will halt for no particular reason. Using the same gcode, it doesn't stop at the same spot. All enable pins for the drivers will go low and the printer will stop moving. The heaters are also disabled. Repetier Host keeps sending gcode commands and the controler keeps responding with ok, as if it was printing normally.
Sending a M122 results in all three of the TMCs to respond correctly . After an emergency stop, the issue goes away until the next print.

I've tried a different usb cable, and lowering current to the motors. My power supply gives out 11.5V but it's still above the minimum 8V the TMCs need.

Am I missing something?
Thanks

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4349676/config.zip)

TouffeTouffe, 2020-03-19:
More testing results:
Using Repetier Host's dry run (no filament extruded) feature with the both heaters off makes the bug go away. But with one heater on, the bug comes back.

TouffeTouffe, 2020-03-19:
My trusty multimeter finally gave me the answer: the printer was doing some quick moves and my (too) weak PSU couldn't keep up. Thus the input voltage dropped to 10.5V which caused the glitch. It has nothing to do with the firmware.

closed 2020-03-19 by TouffeTouffe

thinkyhead, 2020-03-19:
Wow, thanks for the info. I will make a note that this is possible and include it in the Troubleshooting list.

TouffeTouffe, 2020-03-25:
@thinkyhead I saw your post in the troubleshooting list and I think I didn't explain correctly: the heaters are turned off irl but the firmware thinks they are still on. So the temperature decreases and thermal runaway protection quickly kicks in. My bad.