MarlinFirmware/Marlin #17204: MKS_SGEN_L TMC2209 Sensorless Homing

LabBoxPrinter, 2020-03-18:
### Bug Description

No movement is observed when homing either the X or Y axis using sensorless homing. A high pitched whining sound and maybe a mm of movement can be seen at most. 

### My Configurations

Custom CoreXY 
Mother board: MKS SGEN L 
TMC2209 drivers on all motors
DIAG pin is connect to the endstop pin via internal connection
UART Mode via internal connection
TMC2209 Drivers are working and the printer can move correcting and without issue 

### Steps to Reproduce

1. Move tool-head to center of machine
2. Turn on machine 
3. Press home X or Y axis 

**Expected behavior:** Sensorless homing to x_min and y_min

**Actual behavior:** Little to no movement observed, some high pitched whining 

#### I changed the senseless homing sensitivity to 0, 50, and 255 on both the X and Y axis. All three settings had the same effect

KAY3D, 2020-03-19:
has this worked before on older marlin firmware versions? if yes, which versions?

LabBoxPrinter, 2020-03-19:
> has this worked before on older marlin firmware versions? if yes, which versions?

I am not sure. This is my first time building a custom machine. I followed a tutorial for it from Teaching Tech Tips

KAY3D, 2020-03-19:
Can you move the x and y axis manually via LCD?

LabBoxPrinter, 2020-03-19:
Yes, both axis move when commanded using the LCD menu. I have also tried wiring the pin next to what was suppose to be DIAG1 to the minimum endstops, just in case this model of TMC2209 had them reversed, like in some other versions.

LabBoxPrinter, 2020-03-19:
Problem solved. I connected the middle pin of the triangle on the TMC2209 to the endstop.

closed 2020-03-19 by LabBoxPrinter