MarlinFirmware/Marlin #17205: ZONESTAR_LCD

clifo, 2020-03-18:
BOARD ZRIB v3.2
Analog keyboard only responds to the button left
[Configuration.h.TXT](https://github.com/MarlinFirmware/Marlin/files/4351376/Configuration.h.TXT)
[Configuration_adv.h.TXT](https://github.com/MarlinFirmware/Marlin/files/4351377/Configuration_adv.h.TXT)

thinkyhead, 2020-03-18:
Did it work before but then stop working in a new version of Marlin?

woutnieuw93, 2020-03-21:
I have the same issue with 2.0.5, did work on the supplied 1.1.8 firmware from zonestar on my p802qr2

woutnieuw93, 2020-03-23:
I'd like to help you guys get more data, whats needed?

thinkyhead, 2020-03-23:
> whats needed?

- The working Firmware from Zonestar to compare against.
- A proper `ADC_KEYPAD_PIN` value for `BOARD_ZRIB_V20`.
- The output of `M43` showing whether any pin conflicts exist.

clifo, 2020-03-26:
HI
Only when I press the key to the left, the beeper is activated. It responds to commands from the console (Rpetier - Host).
My English is very weak, I apologize

M0101019 -> 1.1.x "bugfix" snapshot date 03152020    - OK
ZONESTAR_Firmware_V3（Marlin_V1.1.8 ) - original by Zonestar - select MODEL_NUMBER        P802QS_5KEY

M020005 -> Marlin 2.0 with bug fixes - date 03192020
[M020005.zip](https://github.com/MarlinFirmware/Marlin/files/4388193/M020005.zip)
[ZRIB_V32_Pinout.pdf](https://github.com/MarlinFirmware/Marlin/files/4388195/ZRIB_V32_Pinout.pdf)
[ZRIB_V3_Schematic.pdf](https://github.com/MarlinFirmware/Marlin/files/4388196/ZRIB_V3_Schematic.pdf)
[ZRIB_LCD_KEYPAY_SCH V4.pdf](https://github.com/MarlinFirmware/Marlin/files/4388199/ZRIB_LCD_KEYPAY_SCH.V4.pdf)
[ZONESTAR_Firmware_V3（Marlin_V1.1.8 )(1).zip](https://github.com/MarlinFirmware/Marlin/files/4388200/ZONESTAR_Firmware_V3.Marlin_V1.1.8.1.zip)
[M010109.zip](https://github.com/MarlinFirmware/Marlin/files/4388203/M010109.zip)

M43 command:

10:29:54.362 : N16 M43*35
10:29:54.362 : PIN:   0   Port: E0        RXD0                                   protected
10:29:54.378 : PIN:   1   Port: E1        TXD0                                   protected
10:29:54.393 : PIN:   2   Port: E4        <unused/unknown>                       Input  = 0                           TIMER3B   PWM:     0    WGM: 1    COM3B: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
10:29:54.393 : PIN:   3   Port: E5        X_MIN_PIN                              protected
10:29:54.393 : .                          X_STOP_PIN                             protected
10:29:54.424 : PIN:   4   Port: G5        SERVO3_PIN                             Input  = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled
10:29:54.440 : PIN:   5   Port: E3        SERVO2_PIN                             Input  = 0    TIMER3A   PWM:     0    WGM: 1    COM3A: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
10:29:54.456 : PIN:   6   Port: H3        SERVO1_PIN                             Input  = 0    TIMER4A   PWM:     0    WGM: 1    COM4A: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
10:29:54.471 : .                          ZRIB_V20_D6_PIN                        Input  = 0    TIMER4A   PWM:     0    WGM: 1    COM4A: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
10:29:54.471 : PIN:   7   Port: H4        FAN1_PIN                               protected
10:29:54.471 : .                          MOSFET_D_PIN                           protected
10:29:54.487 : PIN:   8   Port: H5        HEATER_BED_PIN                         protected
10:29:54.487 : .                          RAMPS_D8_PIN                           protected
10:29:54.503 : PIN:   9   Port: H6        FAN_PIN                                protected
10:29:54.503 : .                          RAMPS_D9_PIN                           protected
10:29:54.503 : .                          ZRIB_V20_D9_PIN                        protected
10:29:54.518 : PIN:  10   Port: B4        HEATER_0_PIN                           protected
10:29:54.518 : .                          RAMPS_D10_PIN                          protected
10:29:54.549 : PIN:  11   Port: B5        SERVO0_PIN                             Input  = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
10:29:54.565 : PIN:  12   Port: B6        <unused/unknown>                       Input  = 0                           TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
10:29:54.581 : PIN:  13   Port: B7        LED_PIN                                Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
10:29:54.596 : .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
10:29:54.596 : PIN:  14   Port: J1        Y_MIN_PIN                              protected
10:29:54.612 : .                          Y_STOP_PIN                             protected
10:29:54.612 : PIN:  15   Port: J0        <unused/unknown>                       Input  = 1
10:29:54.628 : PIN:  16   Port: H1        LCD_PINS_RS                            Output = 1
10:29:54.628 : PIN:  17   Port: H0        LCD_PINS_ENABLE                        Output = 0
10:29:54.628 : PIN:  18   Port: D3        Z_MIN_PIN                              protected
10:29:54.643 : .                          Z_STOP_PIN                             protected
10:29:54.643 : PIN:  19   Port: D2        <unused/unknown>                       Input  = 1
10:29:54.659 : PIN:  20   Port: D1        <unused/unknown>                       Input  = 0
10:29:54.674 : PIN:  21   Port: D0        <unused/unknown>                       Input  = 0
10:29:54.674 : PIN:  22   Port: A0        <unused/unknown>                       Input  = 0
10:29:54.690 : PIN:  23   Port: A1        LCD_PINS_D4                            Output = 0
10:29:54.690 : PIN:  24   Port: A2        E0_ENABLE_PIN                          protected
10:29:54.690 : PIN:  25   Port: A3        LCD_PINS_D5                            Output = 0
10:29:54.706 : PIN:  26   Port: A4        E0_STEP_PIN                            protected
[... cut ...]
10:29:55.065 : PIN:  64   Port: K2 (A10)  BTN_EN1                                Input  = 0
10:29:55.065 : PIN:  65   Port: K3 (A11)  <unused/unknown>   Analog in =   840   Input  = 1
10:29:55.081 : PIN:  66   Port: K4 (A12)  MAX6675_SS_PIN                         Input  = 1
10:29:55.081 : PIN:  67   Port: K5 (A13)  <unused/unknown>   Analog in =   985   Input  = 0
10:29:55.096 : PIN:  68   Port: K6 (A14)  <unused/unknown>   Analog in =   984   Input  = 0
10:29:55.112 : PIN:  69   Port: K7 (A15)  <unused/unknown>   Analog in =  1023   Input  = 1

bcdiaconu, 2020-06-04:
Any news on this? I have the exact same problem.

bcdiaconu, 2020-06-05:
If needed, can find configs for this printer here https://github.com/bcdiaconu/Marlin/tree/2.0.x/Marlin

bcdiaconu, 2020-06-11:
News: If I press center button in combination with up it will enter Menu. If inside menu I press center buttn + down will go down. But it seems to me that center button + up results not only as center button but rather a sequence of buttons.

bcdiaconu, 2020-06-15:
It seems to me that center+up is doing a combination of up+center. center+down is doing a down.

Can you please help with this?

bcdiaconu, 2020-06-15:
Left button does a back followed by down.

I entered in menu -> About Printer -> Board Info and it shows "MKS Gen >0 v1.3" instead of ZRIB

ellensp, 2020-06-15:
P802QS_5KEY uses A10 for ADC_KEYPAD
the file says 

 static const _stADCKeypadTable_ stADCKeyTable[] PROGMEM = {
    // VALUE_MIN, VALUE_MAX, KEY
    { 4000, 4096, BLEN_REPRAPWORLD_KEYPAD_F1 + 1 },     // F1
    { 4000, 4096, BLEN_REPRAPWORLD_KEYPAD_F2 + 1 },     // F2
    { 4000, 4096, BLEN_REPRAPWORLD_KEYPAD_F3 + 1 },     // F3
    {  300,  500, BLEN_REPRAPWORLD_KEYPAD_LEFT + 1 },   // LEFT
    { 1900, 2200, BLEN_REPRAPWORLD_KEYPAD_RIGHT + 1 },  // RIGHT
    {  570,  870, BLEN_REPRAPWORLD_KEYPAD_UP + 1 },     // UP
    { 2670, 2870, BLEN_REPRAPWORLD_KEYPAD_DOWN + 1 },   // DOWN
    { 1150, 1450, BLEN_REPRAPWORLD_KEYPAD_MIDDLE + 1 }, // ENTER
  };

but marlin 2 doesnt do it this way...

ellensp, 2020-06-15:
marlin 2 has

```
#ifndef ADC_BUTTONS_VALUE_SCALE
    #define ADC_BUTTONS_VALUE_SCALE       1.0  // for the power voltage equal to the reference voltage
  #endif
  #ifndef ADC_BUTTONS_R_PULLUP
    #define ADC_BUTTONS_R_PULLUP          4.7  // common pull-up resistor in the voltage divider
  #endif
  #ifndef ADC_BUTTONS_LEFT_R_PULLDOWN
    #define ADC_BUTTONS_LEFT_R_PULLDOWN   0.47 // pull-down resistor for LEFT button voltage divider
  #endif
  #ifndef ADC_BUTTONS_RIGHT_R_PULLDOWN
    #define ADC_BUTTONS_RIGHT_R_PULLDOWN  4.7  // pull-down resistor for RIGHT button voltage divider
  #endif
  #ifndef ADC_BUTTONS_UP_R_PULLDOWN
    #define ADC_BUTTONS_UP_R_PULLDOWN     1.0  // pull-down resistor for UP button voltage divider
  #endif
  #ifndef ADC_BUTTONS_DOWN_R_PULLDOWN
    #define ADC_BUTTONS_DOWN_R_PULLDOWN   10.0 // pull-down resistor for DOWN button voltage divider
  #endif
  #ifndef ADC_BUTTONS_MIDDLE_R_PULLDOWN
    #define ADC_BUTTONS_MIDDLE_R_PULLDOWN 2.2  // pull-down resistor for MIDDLE button voltage divider
  #endif
```

From the circuit diagram I can see two that do not match. So try the following in ultralcd.cpp

```
  #ifndef ADC_BUTTONS_UP_R_PULLDOWN
    #define ADC_BUTTONS_UP_R_PULLDOWN     10.0  // pull-down resistor for UP button voltage divider
  #endif
  #ifndef ADC_BUTTONS_DOWN_R_PULLDOWN
    #define ADC_BUTTONS_DOWN_R_PULLDOWN   1.0 // pull-down resistor for DOWN button voltage divider
  #endif

```

bcdiaconu, 2020-06-16:
With this modification will work center+up and left. down or center+down  will no longer work

ellensp, 2020-06-16:
well that makes zero sense...

above I extracted the tables from P802QS_5KEY
```
{ 300, 500, BLEN_REPRAPWORLD_KEYPAD_LEFT + 1 }, // LEFT
{ 1900, 2200, BLEN_REPRAPWORLD_KEYPAD_RIGHT + 1 }, // RIGHT
{ 570, 870, BLEN_REPRAPWORLD_KEYPAD_UP + 1 }, // UP      
{ 2670, 2870, BLEN_REPRAPWORLD_KEYPAD_DOWN + 1 }, // DOWN   
{ 1150, 1450, BLEN_REPRAPWORLD_KEYPAD_MIDDLE + 1 }, // ENTER
```

I did some math from the values on the circuit diagram

```
resistors in parallel  voltage divider ratio     Expected DAC value             Direction
4.7 + 4.7 = 2.35       2.35/4.7 = 0.5            0.5         * 4096 = 2048      right
4.7 + 1   = 0.82       0.82/4.7 = 0.174468085    0.174468085 * 4096 = 714.6     up
4.7 + 10  = 3.20       3.20/4.7 = 0.680851064    0.680851064 * 4096 = 2788.76   down
4.7 + .47 = 0.43       0.43/4.7 = 0.091489362    0.091489362 * 4096 = 374.7     left 
4.7 + 2.2 = 1.50       1.50/4.7 = 0.319148936    0.319148936 * 4096 = 1307.234  enter
```
And arrived at the same values as above.  (ie values are approximately mid way for each direction range)

bcdiaconu, 2020-06-16:
I've looked on the board and the resistor values are correct. Written on the board itself but also the notation written on smd is right.

ellensp, 2020-06-16:
ok lets attack this from another direction. direct measurement!
@bcdiaconu if I was to write a sketch to dump the ADC values to serial console can you upload it to your controller and tell me what values are displayed when you press each button? 
This will erase your firmware, is that an issue?
Can you/Have you  used serial debug before?

bcdiaconu, 2020-06-16:
yes, we can do that, no problem.

ellensp, 2020-06-16:
this is a tester I wrote for ramps thermitors but will work for this https://github.com/ellensp/RAMPS_T0-T2_tester
change #define TEMP_0_PIN       A13  to #define TEMP_0_PIN       A10 
and monitor at 9600 
take note on what values T0 RAW value says when you hold down each button.

bcdiaconu, 2020-06-16:
No button pressed: 1019
UP: 695
LEFT: 92
BOTTOM: 179
RIGHT: 511
CENTER: 322
-------------- EXTRA Maybe will help with debug -------------------
BOTTOM+TOP: 165
CENTER+TOP: 281
CENTER+LEFT:76
CENTER+RIGHT:245
CENTER+BOTTOM: 129

Unfortunatelly I will have to leave in 10 mins.  But tomorrow I will have a bit more time if needed.

bcdiaconu, 2020-06-16:
Is it a 12bit or 10bit adc? because from no button pressed value it seems to be a 10 bit resolution.

ellensp, 2020-06-16:
yes its a 10bit adc...  on the mega2560

ellensp, 2020-06-16:
Right this is taking to long... not having the Zonestar LCD + keypad. is somewhat limiting. So I built the keypad from stuff I had handy. I tested it and get same results as you do for the raw ADC values.  
[image]

This should speed things up...  but for now its 5am! 
Tomorrow Ill swap the 12864 lcd for a 20x4 tomorrow and get marlin to talk to it.

ellensp, 2020-06-16:
I just got my test system working.  only had to change some pins to make it work on my ramps.
It works seems to work fine.

Not 100% sure what left and right are meant to do..
But center enters menu and selects things, up/down scrolls trough options.
left seems to go up one menu...  whats right meant to do?

One thing I do notice is that I have the ADC_KEYPAD_PIN  set to  4 
But M43 reports "PIN:  58   Port: F4 (A 4)  <unused/unknown>   Analog in =  1023   Input  = 0"

One of you posted a M43 and it had something else on the pin.

So I suspect after all this its just a pin configuration issue.

ellensp, 2020-06-16:
In the file pins_ZRIB_V20.h
after `#if ENABLED(ZONESTAR_LCD)` add these six lines
```
  #undef SHIFT_OUT
  #undef SHIFT_CLK
  #undef SHIFT_LD
  #undef BTN_EN1
  #undef BTN_EN2
  #undef BTN_ENC
```
this should be before   #undef LCD_PINS_RS

bcdiaconu, 2020-06-17:
Yup. That's it. With these undefine in pins_ZRIB_V20.h and marlin's initial ultralcd.cpp printer menu buttons seems have proper functionality. But I get false pushes (double, triple) and I just need to add more debounce time but ADC_BUTTON_DEBOUNCE_DELAY it is at maximum allowed by sanity check. Is there another way or removing the line from sanity check?

bcdiaconu, 2020-06-17:
OK. I guess just modifing limits in sanity check will not do since the variable comparing against is only 8 bit. So, is there another way or shoul i both modify the sanity check and the variable type to a 16bit size?

boelle, 2020-06-21:
@clifo still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

bcdiaconu, 2020-06-25:
Hi. For me the issue was solved by doing the step described here: https://github.com/MarlinFirmware/Marlin/issues/17205#issuecomment-645010241

The thing that remains is that the debouncing is still low even the value of ADC_BUTTON_DEBOUNCE_DELAY if set at maximum. It will double or triple the command.

clifo, 2020-06-25:
The buttons work, but the problem is with the debounce. The problem is also with MARLIN 1.1.9.1 - but irregularly.

thinkyhead, 2020-06-26:
I've added a patch modifying `ADC_BUTTON_DEBOUNCE_DELAY` so you can set any value.

bcdiaconu, 2020-07-11:
@thinkyhead I have created a new bug report regarding denounce. Please see [link to bug](https://github.com/MarlinFirmware/Marlin/issues/18614)