MarlinFirmware/Marlin #17206: [BUG] unstable nozzle movement during bed leveling

emaayan, 2020-03-18:
### Bug Description

when doing g29 ABL , the nozzle would stop at random positions instead at even intervals which eventually woudl cause it to crash into the frame 

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4351509/Marlin.zip)
[Uploading Marlin.zip…]()

### Steps to Reproduce

1. start with the following script 
G28 ; home all axes
M420 Z10 ; set fade leveling at 2mm for BLTouch (OK to send for Non BLTouch)
G29 ; probe bed
2. watch the nozzle begin to probe 

**Expected behavior:** 
nozzle probing at even distances 

**Actual behavior:** 
nozzles moves 2mm stops and probes, them moves 3 mm, probes
but it would behave as though it would move all the distance. 

SKR TURBO 1.4 ,TMC2209, Sensor less homing 
this behaviour started to occur ever since version 2.0.4.4, on 2.0.4.3 it didn't happen 
it also happens on 2.0.5 ,2.0.5.1 

#### Additional Information
https://www.youtube.com/watch?v=-uqVs568Fsw
at 0:08 you can see the short distance move, while 0:15 you can the distance it's meant to move all the time , this causes it to suddenly move to the next row in the MIDDLE of the bed while it thinks it finished the first row .

boelle, 2020-06-20:
@emaayan is issue gone?

emaayan, 2020-06-21:
> @emaayan is issue gone?

no, in fact i believe https://github.com/MarlinFirmware/Configurations/issues/62 is a variant of it, ( and no changing the threshold didn't help)

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle