MarlinFirmware/Marlin #17210: [BUG] Inconsistency between formatting of data reported by M906 and M503

guruathwal, 2020-03-19:
### Bug Description
`M906` and `M503` reports Stepper drivers' current settings in totally different formating. While on the other hand commands like `M92`, `M203`, `M201` have the same formatting as `M503`. This also results in some sort of code duplication for reporting the same data in different formats in Marlin Firmware and Hosts. 

### Steps to Reproduce
1. Send `M906` Command from Host
2. Send `M503` Command from Host
3. Compare the formatting of received data from Step 1 and Step 2 

### Expected behavior:
Data queried through `M906` and `M503` should have similar formatting like:
Through `M906` as:
```
M906 X:400 Y:400 Z:800 Z2:800 E0:800 E1:800
```
Through `M503` as:
```
echo:Stepper driver current:
echo:  M906 X:400 Y:400 Z:800 Z2:800 E0:800 E1:800
```
This may also reduce the `PROGMEM` usage by removing the text "driver current".

### Actual behavior:
`M906` command for the Stepper drivers' current setting reports data the following format:
```
X driver current: 400
Y driver current: 400
Z driver current: 800
Z2 driver current: 800
E driver current: 800
E2 driver current: 800
```
While the same data is reported in a different format when queried through `M503` command under the heading of `M906` Command as:
```
echo:Stepper driver current:
echo:  M906 X400 Y400 Z800 E800
echo:  M906 I1 Z800 E800
```

Jaroslav99, 2020-04-02:
I have successfully connected the display TFT35 v2. But I have another problem. I cannot connect the printer via Octoprint. When attempting to connect, the display shows "Stepper driver current" and the display will beep. I use Marlin 1.1.9. Can you help me solve this problem? Thank you, Yours sincerely Jaroslav

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

guruathwal, 2020-06-24:
I am sorry, I missed the notification of your replies to this issue.

About this issue:
Here are the config files.
[marlin config skr1.4.zip](https://github.com/MarlinFirmware/Marlin/files/4824828/marlin.config.skr1.4.zip)
Although I doubt if configuration files are relevant in this case because the issue is related to the format/structure of the responses sent by marlin firmware, where different formats are used to report the same data.