MarlinFirmware/Marlin #17215: [BUG] Auto Z Stepper Alignment Corrections Not Being Applied

eehusky, 2020-03-20:
### Bug Description
When the following options are enabled 
```c
#define Z_MULTI_ENDSTOPS
...
#define Z_STEPPER_AUTO_ALIGN
```

The ordering of this conditional (line 376 in stepper.cpp) will not allow for the Z_STEPPER_AUTO_ALIGN branch to be selected which prevents the auto alignment corrections from be applied.  
```c
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z_APPLY_STEP(v,Q) TRIPLE_ENDSTOP_APPLY_STEP(Z,v)
  #elif ENABLED(Z_STEPPER_AUTO_ALIGN)
    #define Z_APPLY_STEP(v,Q) TRIPLE_SEPARATE_APPLY_STEP(Z,v)
  #else
    #define Z_APPLY_STEP(v,Q) do{ Z_STEP_WRITE(v); Z2_STEP_WRITE(v); Z3_STEP_WRITE(v); }while(0)
  #endif
```
I'm not sure what the preferred method for addressing this would be but for my purposes i swapped the order of the condition to look like this and it seems to be working and it makes sense (If you have ZSTEPPER_AUTO_ALIGN it implies you have Z_MULTI_ENDSTOPS as well so conditional priority should be given to the more stringent selection of having auto alignment enabled, or you could nest the conditionals).  The same issue also applies to the quad stepper version as well.
```c
  #if ENABLED(Z_STEPPER_AUTO_ALIGN)
    #define Z_APPLY_STEP(v,Q) TRIPLE_SEPARATE_APPLY_STEP(Z,v)
  #elif ENABLED(Z_MULTI_ENDSTOPS)
    #define Z_APPLY_STEP(v,Q) TRIPLE_ENDSTOP_APPLY_STEP(Z,v)
  #else
    #define Z_APPLY_STEP(v,Q) do{ Z_STEP_WRITE(v); Z2_STEP_WRITE(v); Z3_STEP_WRITE(v); }while(0)
  #endif
```

### Steps to Reproduce
1. Run G34 to auto align Z steppers.
2. Run G28
3. Run G34 and note that the initial iteration is still showing uncorrected values.
4. Make the modifications i suggested
5. Re-run steps 1,2,3 and note that the second run of G34 will stop after a single iteration since the corrections are applied. (Expected Behavior)

#### Additional Information
* Based on commit 129b1bb8d4eadf6e24b19ed4ef7f50e45a95e508

sjasonsmith, 2020-03-20:
> ZSTEPPER_AUTO_ALIGN it implies you have Z_MULTI_ENDSTOPS as well

Not always. I use Three Z steppers with zero endstops, just a probe.

eehusky, 2020-03-20:
Ah alright, reason number one i didn't make a pull request for this, way too many variations.

Does that mean you have to run the auto align procedure every time you start up?

Another interesting question, how do the offsets applied by M666 mix in with this?  Would the corrections generated by iteration of G34 and applied to Z2 and Z3 map directly into the end stop adjustments stored by M666?

Neiljclements, 2020-03-30:
I’m using an inductive probe on my CR-10S5 with dual Z steppers and I get similar results.
It moves to each side and takes a measurement, and doesn’t make any adjustments.
I’ve tried putting the axis way out - by 3mm - and it doesn’t correct it.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle