MarlinFirmware/Marlin #17216: Auto Homing goes to Right side but Individual Homing of X goes to left

Talha909, 2020-03-20:
SKR 1.4 Turbo with TMC2130 SPI Sensorless Homing Enable. Marlin 2.0.5.1. I did not installed the end stops. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4358296/Marlin.zip)

I am facing new problem. My individual axis do this when i do individual homing.

X axis go to left

Y axis go to backward

Z axis go to down.

I think these are correct homing for the individual.

But when I click on auto home the printer do this

X axis go to right

Y axis go to backward

Z axis go to down.

What is wrong ?

boelle, 2020-06-20:
@Talha909 is the issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle