MarlinFirmware/Marlin #17217: [BUG] skr mini e3 v1.2 Error USB_EP_TYPE_ISO undeclared here

ellensp, 2020-03-20:
### Bug Description

Issues with library "USBComposite for STM32F1" under platform = ststm32
Effects new users installing platformio. 
When they attempt to compile the library "USBComposite for STM32F1" version 0.97 is installed.
This results in the error "platformio\lib\USBComposite for STM32F1_ID2052\usb_audio.c:225:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function);"  

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4360231/configs.zip)

### Steps to Reproduce

1. Set default_envs = STM32F103RC_btt_512K in palatformio.ini
2. set #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 in Configuration.h
3. set #define SERIAL_PORT -1 in Configuration.h  (needed for this board)
4. force the current version of USBComposite library (platformio will not tell you there is a new version or attempt to get it if you already have an older version) so add USBComposite for STM32F1@==0.97 to lib_deps of env:STM32F103RC_btt_512K

**Expected behavior:** 

It should Compile

**Actual behavior:** 

It errors with 
.platformio\lib\USBComposite for STM32F1_ID2052\usb_audio.c:225:32: error: 'USB_EP_TYPE_ISO' undeclared here (not in a function);

**Additional Info**

If you force PIO to use 0.91 with USBComposite for STM32F1@<=0.91
This will compile, but I think is the wrong way to go.

**What is happening**

The version of framework-arduinoststm32-maple is to old for library USBComposite for STM32F1 > 0.91
The 'proper' fix is to manually update framework-arduinoststm32-maple by doing the following
```
# cd ~/.platformio/packages/
# mv framework-arduinoststm32-maple junk
# git clone https://github.com/rogerclarkmelbourne/Arduino_STM32.git
# mv Arduino_STM32 framework-arduinoststm32-maple
# cd framework-arduinoststm32-maple
# cp ../junk/package.json .    
# cd tools
# cp ../../junk/tools/plat*py .
# rm -rf ../../junk
```

Reported on Discord by Lain
Full compile error log https://pastebin.com/7qxLSR7P 
Discussion and fix of actual issue https://github.com/arpruss/USBComposite_stm32f1/issues/55

Can 'we' automate the updating of framework-arduinoststm32-maple somehow to fix this 'correctly'

qwewer0, 2020-03-20:
> set #define SERIAL_PORT -1 in Configuration.h (needed for this board)

Do you mean `SERIAL_PORT_2` -1?

ellensp, 2020-03-20:
No. Either serial port can be -1, but -1 needs to set somewhere

qwewer0, 2020-03-20:
SKR Mini E3 v1.2 needs SERIAL_PORT set as 2, and SERIAL_PORT_2 set as -1.

ellensp, 2020-03-20:
not just to compile it doesn't. And that is all I'm concerned about here.  The only Conditionals test is that a serial port is -1.

thinkyhead, 2020-03-21:
> Can 'we' automate the updating of framework-arduinoststm32-maple somehow to fix this 'correctly'

No. But someone can submit a patch to the project and when the new version is released we can specify it as the new required version.

thinkyhead, 2020-03-21:
Back to the OP… I believe that if you want USBComposite you should be using either the `STM32F103RC_btt_USB` environment or the `STM32F103RC_btt_512K_USB` environment. Both of these include the `-DUSE_USB_COMPOSITE` flag. The others do not. That said, in my testing your simple configuration changes all the environments compiled fine with USBComposite_stm32f1 up to version 0.96. So, I think it should be safe to specify 0.96 as the required version.

thinkyhead, 2020-03-22:
1674df00b1

closed 2020-03-22 by thinkyhead