MarlinFirmware/Marlin #17218: Neopixel wrong colors

McPaul0, 2020-03-20:
### Bug Description

NEOPIXEL_STARTUP_TEST enabled
So when the power is on it should show colors: Red, Green, Blue and then off.
But it shows Green, Orange, Green, Blue and then Green is always on.
When I set color with M150 command there is no Red color at all.
M150 R255 - Orange (Red+Green)
M150 U255 - Green
M150 B255 - Light blue (Blue+Green)
M150 W255 - Strange white

I've tried two different WS2812B neopixels with the same problem.
The same issue on BTT SKR 1.4, 1.3 and MKS SGenL
When I change any value in the firmware, steps per mm, or the current to the motors and the flash it may help, and the colors get right till the next flash.
That issue was since Marlin2.0

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Enable neopixel in firmware
2. Flash
3. and turn on the Power

**Expected behavior:**
Shows right colors

#### Additional Information
[https://cloud.mail.ru/public/4fUg/3L8og3Dm9](url)

ellensp, 2020-03-21:
No Config files. This make everyone very reluctant to help as they can't try your configs on their hardware.
Most likely you have "#define NEOPIXEL_TYPE   NEO_GRBW" and you don't have the GRBW type. Try "#define NEOPIXEL_TYPE NEO_GRB"
But this is only a guess as I can't see you configuration files!

McPaul0, 2020-03-21:
I've tried all the combinations without success.
I didn't fint the button to attach the conf files. So google drive: https://drive.google.com/open?id=131_gMRp0Bi0bb0DB0fTnWOpSiSBF3cjR

KAY3D, 2020-03-21:
what was the last workingn marlin version for your neopixels?

McPaul0, 2020-03-21:
> what was the last workingn marlin version for your neopixels?

I'd say before september 2019

boelle, 2020-06-20:
@McPaul0 is issue gone?

McPaul0, 2020-06-21:
No. Marlin 2053 the same issue. 
When I enable #define UNKNOWN_Z_NO_RAISE it lights wrong colors.
But when i Disable that feature, it is ok!

ellensp, 2020-06-21:
Tested neopixel (NEO_GRB) on marlin 2.0.5.3 on a ramps with and without UNKNOWN_Z_NO_RAISE.
Couldn't reproduce the issue.  Ill try a skr 1.3 next (don't have a skr 1.4)

ellensp, 2020-06-21:
on skr 1.3 Marlin 2.0.5.3 with NEO_GRB power on test gives Green, Red, Green, Blue. an extra green at the start... can't see any other issues

thinkyhead, 2020-06-21:
> No. Marlin 2053 the same issue.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

---
@boelle @ellensp ☝️ ☝️ ☝️ ☝️ ☝️

ellensp, 2020-06-21:
Still displays 4 test colours on NEOPIXEL_STARTUP_TEST in current bugfix

ellensp, 2020-06-21:
In 
```
void Marlin_NeoPixel::init() {
  SET_OUTPUT(NEOPIXEL_PIN);
  set_brightness(NEOPIXEL_BRIGHTNESS); // 0 - 255 range
  begin();
  show();  // initialize to all off
```

show() with the comment "initialize to all off" is what displays first green colour..  
commenting it out and the extra colour is gone...    looking deeper.

McPaul0, 2020-06-21:
> In
>
> ```
> void Marlin_NeoPixel::init() {
>   SET_OUTPUT(NEOPIXEL_PIN);
>   set_brightness(NEOPIXEL_BRIGHTNESS); // 0 - 255 range
>   begin();
>   show();  // initialize to all off
> ```
>
> show() with the comment [...]

Try to set the colour witn M150 command.

ellensp, 2020-06-21:
"Try to set the colour witn M150 command." I did at ever step of testing...  M150 R32  is red, M150 U32 is green M150 B32 is blue..  using low values to avoid using to much current.  
I suspect you simply set the wrong type of neopixel. There are 36 possible types. see https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L79 for full list.

McPaul0, 2020-06-21:
> I suspect you simply set the wrong type of neopixel. There are 36 possible types. see https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L79 for full list.

May be, But I tried many of them. Not all 36, but 10-15 with no success. I even tried to connect different LEDs. I currently use 4 colors RGBW.

So now I'm afraid to make changes in marlin. For example If I set TMC2209 in speadcylce, or simply change microsteps? then it shows the wrong colors.
I enabled NEOPIXEL_STARTUP_TEST to see, is it good or not right after upload the FW.

ellensp, 2020-06-21:
I would set up an arduino with the standard  Adafruit_NeoPixel and work out/confirm what it needs on simpler hardware with a lot less code to compile...

closed 2020-06-23 by thinkyhead