MarlinFirmware/Marlin #17219: Err: EEPROM VERSION

iMeeHz3D, 2020-03-20:
Hi, i'v been install the new marlin 2.0.5.1 today.

I don't know if i have do an error during the codage, but i have an error when i start my 3D printer ! 
The message is visible in the same place of a thermistance error.

I think it's not annoying for print but i haven't tested. 

I want to know if i'l alone to have this problem ! 

Thank

GMagician, 2020-03-20:
Try with M502 & M500

closed 2020-03-21 by thinkyhead

yonutix, 2020-04-12:
What is M502/M500 ?

ellensp, 2020-04-12:
3D printers 101: They only do what you tell them to do. The language they use is called g-code.
see https://marlinfw.org/meta/gcode/ where you can look up what the gcodes M500 and M502  
You send the printer these gcodes and it does very basic things, move, home, heatup save settings etc

djamelfel, 2020-05-08:
Hello @iMeeHz3D , 
I just made the update too and get the same behavior than you.

Did someone know why my printer is complaining?

djamelfel, 2020-05-08:
Thanks you for the explaination, it's clear and all good :)

thinkyhead, 2020-05-10:
I've just patched `bugfix-2.0.x` EEPROM error handling in a way that (hopefully) improves it.