MarlinFirmware/Marlin #17221: [BUG] (G33 probing failed)

cookthijs, 2020-03-20:
### When using G33(delta autocalibrate) the printhead/probe stops mid-air and starts moving up before giving an error message ''Probing failed'' 

G33 worked fine, after a while it stopped working and started acting like stated above. used to run marlin 1.1.9 and tried to fix it by installing marlin 2.0.5. issue still remained. installed marlin 2.0.x bugfix to get a log. 
### My Configurations
[Configuration.zip.zip](https://github.com/MarlinFirmware/Marlin/files/4361340/Configuration.zip.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [send G33 command]
2. [wait until printer starts moving]
3. [receive probing failed message]

**Expected behavior:** [Delta calibration. (so to probe the buildplate and calibrate the printer]

**Actual behavior:** [printerhead/probe stops after  lowering +/- 9 cm instead of +/- 35 cm, stops, and then moves a bit up after which it starts probing in the air. after that it sends a probing failed error message ]

#### Additional Information
[printerlogfile.txt](https://github.com/MarlinFirmware/Marlin/files/4361358/printerlogfile.txt)
I've added the LOG I got from my printer in repetier host
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jafidod, 2020-04-03:
The ''Probing failed'' message is coming correctly because the probing exceeded "#define Z_PROBE_LOW_POINT" parameter. On the other hand it is just a safety mechanism and the behavior of G33 as I can confirm looks buggy because I am running 2.0.5.3 too and I calibrated my modified Anycubic Kossel manually to reach each point at the bed with error max 1 mm in height. After that I executed G33 and even in such a case the G33 starts doing unexpected moves and the nozzle height goes up and down in range of centimeters which is not the expected behavior. I've looked at code of G33 and some related modules but I cannot find there any obvious bug.

mak0t0san, 2020-04-11:
Did you set the printer height correctly? I ran into this and it was because the height was not correct. I had to increase the height a bit.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle