MarlinFirmware/Marlin #17223: Heating stop before

diego0078, 2020-03-20:
Hi. I have problém with heating bed 
If i set some temp  heating stop 10 degree leds for example set 60*c hotbed stop heating 50 *c
But If i try calibrate PID  heating is ok
Please help 
Marlin firmware 2.0.4 
Skr 1.3

ellensp, 2020-03-21:
You have a problem following instruction too. Please include Configuration files as requested.

diego0078, 2020-03-21:
`/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020004

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 5004 lines cut ...]
#endif

/**
 * User-defined menu items that execute custom GCode
 */
//#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
  #define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN  // Return to status screen after a script

  #define USER_DESC_1 "Home & UBL Info"
  #define USER_GCODE_1 "G28\nG29 W"

  #define USER_DESC_2 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  #define USER_DESC_3 "Preheat for " PREHEAT_2_LABEL
  #define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  #define USER_DESC_4 "Heat Bed/Home/Level"
  #define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  #defi

chgi, 2020-03-28:
This is just a guess, but you seem to assume that your bed uses PID controlled heating.
However, in the config you posted, `#define PIDTEMPBED` is commented out.

PLA7IN, 2020-04-19:
I had this problem too and found this issue. With the conclusion @chgi found, I guessed I just have to do an PID-Tuning after my Update to Marlin 2.0.5.3. and now it works. Thanks @chgi and thanks to @diego0078 . I can only hope this helps for you too.

stavinsky, 2020-05-26:
I solved this problem by resetting eeprom.  On 60C it stop on 50.4C. And pid calibration over 10 rounds

giPeteR, 2020-06-19:
I'm not sure if this is related but, for some reason the code is made to limit max-heating to be -15° below max.

So if you in _configuration.h_ for example set:
#define HEATER_0_MAXTEMP 300

The actual max temp would be 275°C.

See line 599 in Marlin/src/gcode/bedlevel/G26.cpp
    _if (!WITHIN(g26_hotend_temp, 165, (HEATER_0_MAXTEMP **- 15**))) {_

and line 46 in Marlin/src/gcode/lcd/M145.cpp
      _ui.preheat_hotend_temp[material] = constrain(v, EXTRUDE_MINTEMP, **HEATER_0_MAXTEMP - 15**);_

...and a bunch of other places.

Not sure what the purpose is to have a _configuration.h_ saying one thing, and having code doing something else.

AnHardt, 2020-06-19:
Above HEATER_0_MAXTEMP you get a MAXTEMP_ERROR including a stopped machine.
Because our heaters always overshoot the target temperature it needs a margin.

closed 2020-06-19 by thinkyhead

giPeteR, 2020-06-20:
Thanks, that make sense in a way.
It would be nice though if the _configuration.h_ mentioned this. It took me a couple of days extra to figure out why I couldn't get even close to the default 275°C.