MarlinFirmware/Marlin #17224: BABYSTEP_MULTIPLICATOR_Z config setting has no effect in v2.0.5

softfoot, 2020-03-20:
I am running v2.0.5 on a Geeetech i3 pro c with a 2860 based controller and a graphical LCD and BLTouch,

I usually set the above to 10 and in previous versions 1.1.8, 2.0.4 it worked as expected but under 2.0.5 it seems to have no effect (the LCD value just goes up .002 at a time.

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4361655/Configurations.zip)
Dave

studiodyne, 2020-03-21:
it seems to be a compiler issue , i'm searching why , but , if i restart my platform io , the bug disappear
I'm currently trying to find this , the code of the lcd menu is the same of old version

studiodyne, 2020-03-21:
try BABY STEP RANGE

thinkyhead, 2020-03-22:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

DynaMight1124, 2020-03-22:
I've tested the latest bugfix branch as suggested. It does make some difference as now BABYSTEP_ZPROBE_GFX_OVERLAY works, whereas on 2.0.5 it didnt, however for me BABYSTEP_MULTIPLICATOR_Z still doesnt work.

softfoot, 2020-03-22:
On 22/03/2020 09:45, DynaMight1124
      wrote:
    
    
      
      I've tested the latest bugfix branch as suggested. It does make
        some difference as now BABYSTEP_ZPROBE_GFX_OVERLAY works,
        whereas on 2.0.5 it didnt, however for me
        BABYSTEP_MULTIPLICATOR_Z still doesnt work.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/MarlinFirmware/Marlin/issues/17224#issuecomment-602172053",
"url": "https://github.com/MarlinFirmware/Marlin/issues/17224#issuecomment-602172053",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
    Confirmed
    Dave

boelle, 2020-06-20:
@softfoot is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle