MarlinFirmware/Marlin #17225: [BUG] Fine tune mesh.  Switching Nozzle servo detached, no reference.

c3D-Dan, 2020-03-21:
### Bug Description
Servo for switching nozzle is deactivated during Fine tune Mesh operation as selected from the UBL menu.  This means no suitable reference point (nozzle to bed) can be used for confirming or fine tuning the mesh.

### My Configurations

[Marlin-conf.zip](https://github.com/MarlinFirmware/Marlin/files/4362206/Marlin-conf.zip)

### Steps to Reproduce

From LCD - Motion > UBL > Step by Step > Fine Tune Mesh

**Expected behavior:** Nozzle servo to remain in either T0 or T1 position throughout the fine tune process

**Actual behavior:** Nozzle servo remains active for the first mesh position.  Upon confirming height with encoder, Z raises, Nozzle Servo deactivates, gantry moves to next grid position.  As nozzle servo has deactivated, there is now no useful reference for fine tuning (distance between nozzle and bed). 

#### Additional Information

As soon as a blocking move is called, servo deactivates.  As a test, by commenting out all blocking moves in ubl_g29.cpp, servo remains active.

boelle, 2020-06-20:
@c3D-Dan is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle