MarlinFirmware/Marlin #17226: [BUG] UBL Mesh Edit Z Motor disables

c3D-Dan, 2020-03-21:
### Bug Description

Motion > UBL > Mesh Edit :   Printer moves to first position (0,0) to edit, then the Z motor disables causing the hotend to crash to the bed

### My Configurations

[Marlin-conf.zip](https://github.com/MarlinFirmware/Marlin/files/4362409/Marlin-conf.zip)

**Expected behavior:** [What you expect to happen]
Z Motor remains enabled, maintaining the Z Positio 

**Actual behavior:** [What actually happens]
Z Motor disabled, Z Position lost. 

#### Additional Information

Printer has belt driven Z (not delta) so nozzle crashes to bed when this occurs.

c3D-Dan, 2020-03-21:
Appears to have something to do with _lcd_hard_stop(); in menu_ubl.cpp 
Commenting this line fixes the problem.  Unintended consequences will no doubt result.

boelle, 2020-06-20:
@c3D-Dan is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle